prusa3d/PrusaSlicer #7041: Adaptive bridging infill orientation to achieve shortest bridging distance

laukejas, 2021-10-01:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S (irrelevant)

### Behavior
When bridging gaps on cylindrical objects, the orientation of bridging infill lines (Print Settings->Infill->Bridging angle) can never be correct for the whole circle, because that angle is always the same for the whole circle (in this screenshot - blue lines):

[image]

This causes inconsistent and unnecessarily extended bridging distance in some places on the circle. There is currently no setting that allows to make this bridging infill radial to the center of the circle (in this second screenshot - yellow lines illustrate the ideal bridging orientation for this specific example):

[image]

If there were an option to set the bridging infill to "dynamic"/"shortest distance" instead of user-specified angle, it would solve this and similar cases,  as it would allow PrusaSlicer to choose an optimal angle for every infill line that provides the shortest bridging distance on any specific spot on the layer.

#### Project File (.3MF) where problem occurs
Attaching example project. See layer 4 for the issue.

[test print.zip](https://github.com/prusa3d/PrusaSlicer/files/7269494/test.print.zip)

RetireeJay46, 2024-05-23:
I agree with this issue and hope that someone can work on it.

flesser, 2026-09-02:
I don't think this feature request should be closed unless it gets either implemented or declined for some reason by a human developer. So here's your "new user comment", dear bot!