prusa3d/PrusaSlicer #7042: M1 fails to stop printer

StevenInc, 2021-10-02:
### Version
Version 2.3.3.+x64

_Use `About->About PrusaSlicer` for release versions_

### Operating system type + version
MAC OS  Version 11.4

### 3D printer brand / version + firmware version (if known)
MK3s with firmware: 3.10.0-448

### Behavior
* _Describe the problem_
* The following code is used to pause the MK3s after the bed leveling to insert a glass bed.
* The code waits for the glass plate to reach temperature and then waits for user input to press the control button.
* Note: This code works perfectly on MK3s machines with 3.9.3-3556
* 
* _Steps needed to reproduce the problem_
copy the following gcode into the Start G-code field
M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.9.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S50 ; set bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M300 S440 P500; gcode start sound to alert ready for glass plate.
M300 S880 P500
M300 S440 P500
G0 F4000l
G0 Z140 Y210 Raise the print nozzle out of the way and Move table forward
;G4 S300 ; pause for 5 mins ;Allow glass to heat to temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M300 S880 P500; gcode start sound to alert ready for first layer adjustment
M300 S440 P500
M300 S880 P500
M300 S440 P500
M117 Press button to continue; LCD Message
M1; user stop wait for user press of the control button.
M117; //reset LCD
G1 Y-3.0 F4000.0 ; go outside print area
G1 Z4.0 ; reset z at thickness of alternative bed
G92 Z0.0
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-10-02:
This is not PrusaSlicer related issue. Please open it at prusa3d/Prusa-Firmware repository. I believe it's intended change, since M1 is not a pause command. 

Closing here, not PrusaSlicer related issue.

closed 2021-10-02 by rtyr

rtyr, 2021-10-02:
I know FW guys did some intentional changes about M1, but I am not completely sure it works as it should, I will ask them about it.

Anyway, M0 should work for you now.

rtyr, 2021-10-02:
FW devs confirmed, that current behavior is correct.

See updated description of M1 command here https://reprap.org/wiki/G-code#M1:_Sleep_or_Conditional_stop.

StevenInc, 2021-10-02:
Thank you,
much appreciated.