prusa3d/PrusaSlicer #7044: Feature Request: Infill Line Multiplier (at least for Grid infill)

thwerks, 2021-10-02:
Multiple line infill is superior to that of single line infill; in strength, in speed and materiel usage.

Infill lines in parallel makes for a stronger part. It is more advantageous to have multiple infill lines in parallel than it is to increase the infill percentage. Individual small infill lines are weak even if you have more of them. It's also faster to print multiple infill lines in parallel than it is to just increase infill percentage. And, multiple infill lines  creates a stronger part for an overall lower weight because the strength gained from lines in parallel is greater than that of just adding more lines.  

 I am aware that Prusa Slicer has Extrusion Width modifier; this is not the same thing and does not  produce the same effect. And it would be significantly harder to tune this for the purpose of making stronger infill. You would be pushing way more filament through the nozzle than its intended for and that adds a significant degree of complexity and undesirable effects to your profile tuning. Possible under extruding, adhesion issues, speed re-calibration, nozzle collisions, inadequate hotend heat , improper filament temperature and many issues come from pushing more filament than the nozzle is intended for. With an infill multiplier all these caveats are eliminated, tuning changes are unnecessary. 

Personally I feel the lack of this feature gives professional FDM manufactures a valid reason not to use Prusa Slicer for some prints; because in some cases you wouldn't be getting as strong, as light, as quickly or as materiel usage efficient a part as you could have with a different slicer. And, when you make parts professionally for customers in the thousands, it really matters!

bubnikv, 2021-10-04:
@thwerks Which slicer support such feature?

Alpacad, 2021-10-04:
@bubnikv I think he's referring to Cura

**Source:** Screenshot of Cura 4.11.0

[image]

phazei, 2024-03-27:
This also seems to have been requested again more recently here:
https://github.com/prusa3d/PrusaSlicer/issues/10585

I'd like to second this request.  It adds a great deal of strength at lower infill percentages.  I realize that if something is at 20% and you choose to multiply the lines it's doubling the amount of infill, but filament is so weak when it's a single layer, it's so much stronger when just a few layers are side by side.  We can choose to increase the perimeter width because it increases strength so much, but just the same, so would an infill multiplier.  Particularly with low infill percentages.  Lower percentages are basically worthless since it'd be so flimsy, but with a multiplier, it could be much stronger.  Here's a gyroid example at 6% infill
Without a multiplier it looks like it would barely stand up:
[image]

But with a 3 multiplier it's much more solid:
[image]

profOnno, 2024-03-29:
This would be nice. I'm very new to slicer programming, so the next screenshot is a quick dirty hack in prusa slicer.
[image]
[image]

I don't know if I have time to make a non hacked version with gui options. Now I changed `FillGyriod.cpp`. In `_fill_surface_single` I've used `make_gyroid_waves` a second time and appended this to the polylines. I've added a argument to the `make_gyroid_waves` method named `offset`.

So far my 2 cents.

Cheers,
Menno

profOnno, 2024-03-30:
Results:
[image]
[image]

My implementation is needs an update, using [ClipperOffset](https://angusj.com/delphi/clipper/documentation/Docs/Units/ClipperLib/Classes/ClipperOffset/_Body.htm). Now the lines are not offset equally (can be a feature, stronger?). 

I'm not sure where howto get the `nozzle width` in the `FillBase`, so hard-coded it for now.
Used the `Maximum length of the infill anchor` as multiplier property (seems to work, still need to figure out, how to add a new property). 

Adding another thought:

Making infill a plugin, makes it easier to fiddle around with infill concepts.

The context is esthetics of flat surface without top and bottom layer (or part of print with modifier). Seeing my result, I want more options on the gyroid, like height scaling flip offset. 

On the gyriod part, horizontal or vertical scaling properties would be nice. When you have a mechanical piece and you know there is a load in one direction.

A text infill would be nice see: [ConnectedText](https://blog.cuttle.xyz/how-to-create-connected-text-for-laser-cutting-with-cuttle/).

The plugin should have a 'get_properties' function, so you can have infill specific properties.

profOnno, 2024-04-01:
Hmm, the shifting kind of works. But the line distance is not equal... So probably need to use ClipperOffset.
I did build it only in the `Gyriod` fill, and should move it to the `Fill.cpp` so it will work for all infills. Also was thinking to go 'down the rabbit hole' and select infill for the infill, so you can have `Gyriod sides with an offset, filled with crossline` infill.

For people interested, [multiline infill](https://github.com/profOnno/PrusaSlicer/commit/90862360de111571272ebbb7b8e0f89f8f841805). You have to build it yourself.

RF47, 2025-05-04:
In case anyone’s interested, I’ve implemented multi-line gyroid infill in Orca Slicer, building on @profOnno  original implementation. My main contribution is an improved algorithm that accurately generates the wave pattern offsets.
https://github.com/SoftFever/OrcaSlicer/pull/9432

[image]

phazei, 2026-09-04:
I still want this feature, it's not a bug, it's a feature request.

Alex20129, 2026-09-09:
I’ll support this as a feature request.

Jan-Soustruznik, 2026-09-09:
Hello, I have moved this feature request to GitHub Discussions in accordance with our [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy).