prusa3d/PrusaSlicer #7045: Feature Request: "Single Extruder Multi-Material" mode should allow different nozzle diameters!

thwerks, 2021-10-02:
I have and Ender 5 Plus and I have two Cyclops hotends mounted to the X carriage. Meaning 4 different extruder motors feed filaments into 2 hot ends, each having 1 nozzle. 

I have set Single Extruder Multi-Material mode. My 2 hot-ends have different nozzle diameters for fairly obvious reasons. But, Prusa Slicer throws a hard warning saying "This is a single extruder multi-material printer; diameters of all extruders will be set to the new value. do you want  to proceed?". Can't this be a soft warning? I can think of a few different reasons why you may want to set a different nozzle diameter. 

And, if I deselect Single Extruder Multi-Material mode, than i loose the ability to set single extruder Multi-Material parameters which are needed.

Please either allow more granular configuration by allowing the user to map which extruder is assigned to which hot end and if that particular hot end is Multi-Material or not. Because fore example someone may have a high temp hot end and a Multi-Material hot end on one carriage for example. A bonus  feature would be to have logic that determines what level of retraction is need; because switching filament on one hotend to a filament on a different hot end is different than switching filament on the same hotend. At least please just allow different diameters. 

Alternatively, just remove the "extruders will be set to the new value" requirement altogether or just make it a soft warning.

bubnikv, 2022-01-20:
> My 2 hot-ends have different nozzle diameters for fairly obvious reasons

Would you please expand on how do you plan or you are already using the 4 extruder setup, each extruder with a different nozzle diameter? What benefits are you getting? What scenarios did you already test? Which work and which do not work?

thwerks, 2022-02-03:
hi, thank you for the reply. i cant test any scenarios because there is a hard error prohibiting the use of two different sized nozzles; please see the previously mentioned error. 

My use case is that i have several different projects that need different sized nozzles during a single print. For example, im making a name plate that has a thick back plate in white PLA for example. a 1mm nozzle is ideal because the backplate doesnt need detail and would be nice to print it as fast as possible. However the letters that are attached to the backplate (also PLA) need finer detail and a .4mm or .6mm would be better.  

i have other projects that would also benefit from being able to use different sized nozzles during a single print but this is the best example i can think or. Please consider making the "extruders will be set to the new value" a warning rather than a hard rule. Thank you

thwerks, 2022-02-07:
each extruder is a 2 in 1 configuration. so 2 extruders = 2 nozzles = 4 total inputs. One nozzle is .4mm and the other is 1mm. The benefit is that a 1mm nozzle is MUCH faster than .4mm obviously but opposingly a .4mm nozzle has more detail. 

I cannot test any scenario where two different sized nozzles are specified because there is an error.   

"This is a single extruder multi-material printer; diameters of all extruders will be set to the new value. do you want to proceed?"

closed 2026-09-29 by prusaslicer-bot[bot]