prusa3d/PrusaSlicer #7047: File Name When exporting file after slicing

mitchcecg, 2021-10-02:
Aplha 2
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
does not 
matter for this

### Behavior
* _Describe the problem_ 
File name to export after slicing defaults to last export file name rather than current object on build plate that is sliced. It would be great if the default was the current object on the build plate. and yes I know you can change it but IMHO it shuld default to the current object.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request? Y

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fe60, 2021-10-05:
> IMHO it should default to the current object.

What exactly do you expect?

Default name:
`{input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode`

For me `{input_filename_base}` gets replaced by the name of the first object shown in the object list of the right panel.

rtyr, 2021-10-05:
I guess you are talking about project name (for example after loading 3MF and then deleting the models). If you don't want the name of the previous project, use `File -> New project (CTRL+N)`.

closed 2026-09-29 by prusaslicer-bot[bot]