prusa3d/PrusaSlicer #7048: Feature Request: Fuzzy skin 2.0 mekyo edition

mekyo1, 2021-10-02:
### **Feature request**

**My idea is:**

Fuzzy skin on outer wall only
using the EXTRUDER so your printer doesn't shake to bits!!

eg of parameters:
**Pulse Frequency per 1mm:**  10 (pulses per 1mm) ppmm
**Pulse swing per pulse:**
**low:** 30%
**high:** 140% 

usually there's no such thing as a new idea under the sun but in the case this is a "new idea"
I request the principle behind this feature be open for everyone to use freely
-mekyo xoxo not all heros wear capes

bubnikv, 2021-10-04:
Don't understand what is different.

> Fuzzy skin on outer wall only

this is already happening

murk-sy, 2021-10-04:
Over/underextruding rather than moving the nozzle sounds interesting but it might lead to blobbing/holes in the outer perimeters. Would like to do a test print with it at least, see if it works well enough.

mekyo1, 2021-10-04:
> 
> Don't understand what is different.
> Fuzzy skin on outer wall only
> […](#)

Current fuzzy skin moves X and Y axis- causing printers to perform lots of acceleration and deceleration and vibrate large heavy gantry and or print-beds and exciting printer resonance - This results in slower prints and wear and tear on printers

instead i suggest pulsating the extruder in a fashion between say 60-110% to create a fuzzy texture on the outerwall -

closed 2026-09-29 by prusaslicer-bot[bot]