prusa3d/PrusaSlicer #7049: Can't open new instance in 2.4.0a3

rkarlsba, 2021-10-02:
### Version
2.4.0a3

### Operating system type + version

macOS 10.15.7

### 3D printer brand / version + firmware version (if known)

Ender 3

### Behavior

Press shift+cmd+i and wait for a new instance to appear - it doesn't. Look in the menus for the choice - it's gone. This has been around for a while and I'd love to get it back

#### Project File (.3MF) where problem occurs

Irrelevant

bubnikv, 2021-10-04:
Is it an ARM based (M1) macbook, or an Intel based?
There seem to be difference in behavior in regard to allowing multiple
instances of the same application to run.

rkarlsba, 2021-10-04:
Intel MacBook Pro (16-inch, 2019)

bubnikv, 2021-10-04:
Don't you have the "single instance" enabled in preferences? If so, the
menu item to start the new instance is grayed out.

rkarlsba, 2021-10-04:
It was there, yes, "Allow just a single PrusaSlicer instance". Works without that ticked. Why is that choice in there anyway?

bubnikv, 2021-10-04:
Single instance of an application is the default behavior on OSX platform.

bubnikv, 2021-10-05:
> > Single instance of an application is the default behavior on OSX platform.

> sorry, but why is this?

Good question, but you have to ask at Apple.

rkarlsba, 2021-10-05:
> > > Single instance of an application is the default behavior on OSX platform.
> 
> > sorry, but why is this?
> 
> Good question, but you have to ask at Apple.

They don't write PrusaSlicer

neophyl, 2021-10-05:
I believe the comment was that the default for OSX programs in general is a single instance, which is a choice made by Apple.

rkarlsba, 2021-10-05:
I'm aware of Apple's defaults. They don't include stuff like allowing more instances open by ticking off a checkbox in their own apps. So my question wasn't about the defaults of Apple, but why this wasn't enabled in the first place. I can't think of a single reason to disable it. Yes, if you're really low on memory, having lots of instances open will be a problem, but nothing will start them for you if you allow for them to be opened. They still have to be created by a keystroke or meny choice. So thus the question.

edwardhorsford, 2021-10-05:
Apple might not commonly have single 'instances' - but nearly all applications can have multiple windows. I'm sure it's been considered, but it would be great if each slicer window just behaved like other windowed apps on MacOs rather than separate instances.

ulab, 2023-01-03:
I have just stumbled over this after complaining on Twitter about not being able to have multiple projects open at once. People then told me to just open multiple instances, which I was not able to until I found the setting in Preferences.

I'd have expected it to just open a new window when I opened a new project or file, as all other apps do. But since it always just asked me to save and close the old project, I just accepted that it was not able to.

So you can guess, I was quite surprised when I learned that there is a way to do just that.

edwardhorsford, 2023-01-03:
Related to @ulab's comment, when opening a saved project and you've already got one open, it would be great if one of the options was 'open in an new instance'. I nearly always want to open them in new instances. 

Right now I frequently forget, so have to cancel the dialog, then open a new instance, then open the project in the new instance.

closed 2026-09-29 by prusaslicer-bot[bot]