prusa3d/PrusaSlicer #7051: Gap between 2 surfaces on a hinge with 2.4 Alpha 3

parhelia, 2021-10-03:
### Version
PrusaSlicer-2.4.0-alpha3+x64-202109301500

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Big Sur Version 11.6 (Build 20G165)
Total RAM size [MB]: 34,360MB

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
A ratchet sliced in PrusaSlicer 2.3 had no issues, but PrusaSlicer 2.4 alpha 3 produced the following defect (please refer to screenshots). **There is a gap where the hinge is supposed to be within the model.**

[image]
[image]

#### Project File (.3MF) where problem occurs
3MF file uploaded
[ratchet-all-in-one-25-mm-dull-teeth.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7273045/ratchet-all-in-one-25-mm-dull-teeth.3mf.zip)

rtyr, 2021-10-03:
There are some errors in the model, it slices ok with repaired model. The difference between 2.3.0 and 2.4.0 is likely due changes described below.

From 2.4.0-alpha1 release notes:

Slic3r, Slic3rPE and PrusaSlicer extensively used admesh library for internal storage and manipulation of triangle meshes. The admesh library was designed with the purpose of importing and fixing errors in STL files, while in Slic3r / PrusaSlicer admesh was also misused as a heavy weight general triangle mesh representation and the same triangle mesh was also kept in memory using a simpler redundant "Indexed triangle set" representation. Starting with this release, PrusaSlicer only uses admesh for loading and fixing STL files, thus the triangle meshes are only kept using the "Indexed triangle set" structure, which reduces memory consumption. **Triangle meshes imported from 3MFs, AMFs and OBJs are no longer repaired by admesh. For PrusaSlicer generated 3MFs it means that the triangle meshes will be loaded exactly the same way they were stored, while earlier PrusaSlicer may have modified a mesh with non-manifold edges slightly by trying to repair it each time a 3MF was loaded. For 3MFs not produced by PrusaSlicer and for AMFs and OBJs it may mean that model errors will not be fixed, however in contrary to the STL file format, the 3MF, AMF and OBJ file formats are indexed triangle sets with triangle neighborship clearly expressed by shared vertex indices, thus cracks in triangle meshes are less likely. Not repairing 3MFs through admesh considerably speeds up loading of PrusaSlicer project files.**

rtyr, 2021-10-03:
Repaired model.

[fixed_ratchet-all-in-one-25-mm-dull-teeth.zip](https://github.com/prusa3d/PrusaSlicer/files/7273140/fixed_ratchet-all-in-one-25-mm-dull-teeth.zip)

parhelia, 2021-10-03:
Thank you for the detailed explanation and solution. Will definitely have to check the sliced output carefully henceforth.

bubnikv, 2021-10-04:
What was the source of the original model? What was the producing
application?

rtyr, 2021-10-04:
It looks like it's generated by Cura.

bubnikv, 2021-10-04:
@rtyr 's analysis is correct. The 3MF was produced by Cura, PrusaSlicer starting with 2.4.0-alpha2 no more tries to fix 3MFs on loading.

We will watch out for similar issues and it may be, that we will re-enable mesh fixing for 3MFs not produced by PrusaSlicer.

BastR, 2021-10-04:
I have a similar problem with a 3mf generated in Fusion360:

[image]
[image]

[Tete.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7280797/Tete.3mf.zip)

bubnikv, 2021-10-05:
The main body - the Body4 model is broken. We have a bug in alpha4, where the "broken" state is not shown, we will fix that.

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]