prusa3d/PrusaSlicer #7058: Please make reload part from disk more open to read same objects with different names

Doprintityourself, 2021-10-04:
### Version
all

### Operating system type + version
Windows1064

### 3D printer brand / version + firmware version (if known)
12 Different brands
###  Wanted Behavior
I am a CAD Devoloper and develop my own products. therfore during product developement I have
a lot of evolving parts. To track the devopment phases in CAD and STL the part files are named
V.1.0.Bracket.PrusaMini.obj
 V.1.1.Bracket.PrusaMini.obj
V.1.3. Bracket.PrusaMini.obj

**Now I like to use reload from disk but it only let me pick whenever asked for the same named file.**

So if I evolut through the design changes the numbers rise and therefore the datafile cannot be picked any more,
even if small changes I have to delete the old and start from scatch, that is often a huge loss off time if I have a lot off helper geometrie applied Variable Radius .height modifier ......

So please it would be very helpful to just be able to load Variants of files with reload form disk.
As I am a professional CAD engineer I even would go further and let the system check if there has been safed a new revision.
so if there is a V.1.0Bracket.PrusaMini.obj in a Prusa slicer and I add a V.1.1Bracket.PrusaMini.obj to the same directory I would get a dialog box, that showes that there is a new version of it and should it be exchanged, vice versa I should be able to selct during Reload from Disk every revision of the changed Part.
Would really appreciate  to streamline workprocesse in Prusa to a more professional workflow

Thank you very much in advance
  
_Is this a new feature request?_
Yes if not already in alpha 3

murk-sy, 2021-10-04:
Right click on the model list -> `Replace with STL` does this and it retains rotation as well.

Function should probably be renamed and moved just under "Reload from disk" though.

Doprintityourself, 2021-10-04:
does not work for me if I assum as soon as you have like in my case add Parts , then the option is greyed out.

But that its what I am looking for not lossing all of the modifications and replace just the part in the same orientation and status, whatever I have done to it.
[image]

Doprintityourself, 2021-10-04:
and additional even if done after relaod it comes in the wrong orintation
[image]
Sometiems I have lots of parts on the Heatbead needed to to a lot of work to organize and print in the same orientation. 
I lost so much time in doing things again and again , sometimes der are up to 7 devlopement steps with x Parts, you can estimate how much time you waste on doing stupid repeting work

murk-sy, 2021-10-04:
I can't reproduce the issue in 2.4.0-alpha3 with .stl/.3mf files, so either obj is treated differently or your export is changing the origin/orientation on every version. Loading different files moves the model around (but retains rotation/scale) while loading versions of the same model (assuming it wasn't moved in base drawing before export) loads it perfectly in place.

Doprintityourself, 2021-10-05:
Ok thank you, I work with a professional CAD product and will investigate it about the orientation topic,
until than thank you for the information

NKT00, 2021-10-30:
Interestingly, I've noticed that "Reload from disk" doesn't seem to work on my machine either, using the latest version. I will test the feature in the next version. As DoPrintItYourself says, it's a handy thing to be able to do.

bubnikv, 2021-12-08:
@NKT00 any news on what is not working yet?

NKT00, 2021-12-08:
Will look tonight. I'll check the version number first and try to replicate.

closed 2026-09-29 by prusaslicer-bot[bot]