prusa3d/PrusaSlicer #7059: [Feature request] Allow for custom upload folders per printer

drumbo, 2021-10-04:
### Version
2.4 Alpha 3

### Operating system type + version
OSX 10.15

### 3D printer brand / version + firmware version (if known)
MK3S+ and MMU2S using 3.10.0 and 1.0.6

### Behavior
Would it be possible to have the capability within the new colour painting feature that allows one to paint straight lines?

So for example if I had a (flat) surface and I wanted to paint two parallel lines at the moment ones needs to do it free hand - it would be nice to be able to ensure the lines are clean and straight.
Currently:
[image]

Feature:(you can see the current paint vs (a rather poor example) what would be nice
[image]

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs

dartrax, 2021-10-05:
This would also be useful for seam painting, like mentioned in
#7035 
There is also a workaround using the num pad mouse control.

My suggestion/FR:
Use the shift key to lock the mouse in a way that it can only be moved up/down or left/right. Like it works when you move objects in MS Office.
I know that currently, Shift is used to delete what was painted before. I suggest using CTRL for that purpose. That is intuitive, because CTRL is already used in list boxes to delete the selection of the clicked item.

dartrax, 2021-10-31:
> ...change the painting tools to NOT paint parts that are "Clipped". This means if you want a clean color change at a layer you can look top-down and then clip to the exact layer, then do a broad sweep with the brush. This would also make straight lines at certain angles possible by resetting the clipping angle.

This part was implemented with 2.4.0 beta 1.

bdwilson, 2022-09-15:
> This would also be useful for seam painting, like mentioned in #7035 There is also a workaround using the num pad mouse control.
>
> My suggestion/FR: Use the shift key to lock the mouse in a way that [...]

FYI, you can also accomplish this on mac using Mouse Keys: https://support.apple.com/guide/mac-help/change-alternate-control-methods-preferences-mchl7b437b35/12.0/mac/12.0.  Configure option key 5x times to turn it on, select the starting point with m, use k to go down (https://support.apple.com/guide/mac-help/control-the-pointer-using-mouse-keys-mh27469/12.0/mac/12.0)

murk-sy, 2023-02-19:
## Workaround

Another workaround using Screen Ruler (from Sourceforge), somewhat janky of course

1) Start ruler
2) Right click on ruler and select Stay On Top and Orientation>Vertical
3) Position the part so the line will be drawn along the side of the ruler
4) Start painting next to the ruler on top, then, without releasing the mouse button, move the mouse over the ruler to the lower point where you wish to paint (see red line for mouse movement on image)
[image]

## Proposed functionality
I wouldn't even mind if the painted line would be cast as displayed on screen, rather than somehow aligned to the model.

Expected behaviour would be something like:
1) Activate line mode (maybe by pressing and holding for 2 seconds without moving the mouse?)
2) First point is painted, and a virtual line is displayed when you move the mouse around
3) when clicking the second time, the on-screen line is painted
[image]

lukasmatena, 2024-02-19:
Closing as a duplicate of #5970.

closed 2024-02-19 by lukasmatena