prusa3d/PrusaSlicer #7060: Using Raft and the 1st Layer of Object

schuessm, 2021-10-04:
### Version
2.3.3

### Operating system type + version
Linux (Ubuntu 20.04 LTS)

### 3D printer brand / version + firmware version (if known)
AM8 DUAL with Klipper

### Behavior
When I use raft, all settings for the first Layer of the object are lost. Then this settings are used for the first raft layer. I've no chance to make a slow first layer on the top of raft. I think it is better to have separated settings for raft for speed an flow an so on. And the main settings of spped and flow are used for slicing the model.
In my case after the raft the first layer of model was print with default speed of 80mm/s and so the layer sticks not really good on the top of raft and the width is to big.

#### Project File (.3MF) where problem occurs
[Testfile.zip](https://github.com/prusa3d/PrusaSlicer/files/7278812/Testfile.zip)

rtyr, 2021-10-05:
Raft was reworked in 2.4.0-alpha1, you can set separate speed and acceleration for raft there. Also bridge flow is no longer used, so surface quality is better.

Latest alpha https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha3

Implemented. Closing.

rtyr, 2021-10-05:
BTW. In you case, it is not 80mm/s, but 75mm/s set for bridging.

closed 2021-10-05 by rtyr

schuessm, 2021-10-05:
Your are right I found it aout in alpha 3 2.4.0 it works fine, I hope all the features there will also be released. Good work. Thank you!