prusa3d/PrusaSlicer #7061: Can't drag multiple .3mf files at once, works with .STLs

erlis, 2021-10-05:
### Version
Version 2.4.0-alpha3+win64

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Prusa MK3s 

### Behavior
While dragging more than one .3mf file from windows explorer only one is added to the Slicer, on the other hand, dragging multiple STL works as expected.... I really don't know if this is an issue inside the Slicer or if it's Windows doing something weird

combolek, 2021-10-05:
Maybe this is intentional? `.3mf` files are complete projects with all settings included. How would that work with multiple `.3mf` files? Which settings should be used?

erlis, 2021-10-05:
but if I drag one at a time then all the models are included on the prusa slicer. I'm expecting to select 4 models from the explorer drag it and end up with the same result of dragging one, then dragging another, then another... 4 times...

timur-tabi, 2021-10-07:
If you drag one at a time, each .3mf file overrides the settings of the previous one.

If you drag multiple at a time, PrusaSlicer can't know which one to use.

neophyl, 2021-10-07:
I thought adding/importing a 3mf just loaded the models and ignores the settings while Opening a 3mf loads in the models and the settings ?  In which case dragging in multiple 3mf files that is just loading in model data wouldn’t have a [Robles with choosing settings.

closed 2026-09-29 by prusaslicer-bot[bot]