prusa3d/PrusaSlicer #7077: Prusaslicer 2.3.3 not slicing single layer objects correctly.

Malaynka, 2021-10-08:
### Version
_Version 2.3.3+win64
_Build:     PrusaSlicer-2.3.3+win64-202107161027

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,115MB

### 3D printer brand / version + firmware version (if known)
Prusa MK3S
### Behavior
When slicing a single layer object, Prusaslicer will only treat the layer as an "infill layer" and therefore the layer cannot be ironed. This report is not about ironing, but came about while working on a project that needed single layer objects to be ironed

**Steps to recreate**
_ create an object only .2mm high.
_ slice it and observe in he preview view that it show fill as 'internal infill" and not "top solid infill"
_ setting top layer to any number except zero and bottom layer to zero has no effect.
**or**
_ load attached file into Prusaslicer and slice
[ironing test.zip](https://github.com/prusa3d/PrusaSlicer/files/7308593/ironing.test.zip)

**Expected Results_**
Single layer objects should be able to be treated as a "top solid infill" layer by user instead of as an "internal infill" layer.

**Actual Results_**
Single layer objects are treated an "internal infill" layer. The .2mm square on the left in the attached image shows no top layer while the .4mm on the right does.one does
[image]

The attached ironing test.3mf  demonstrates the results of a .2mm object and a .4mm object being sliced with ironing enabled

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[ironing test.zip](https://github.com/prusa3d/PrusaSlicer/files/7308595/ironing.test.zip)

bubnikv, 2021-12-08:
Tested with PrusaSlicer 2.4.0-beta3, where some ironing issues were ironed out, I see no issue.

Would you please verify with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-beta3 ?

Malaynka, 2021-12-11:
I downloaded beta3 and loading in the same .3mf file above. A single layer object is sliced as "solid infill layer" instead of a "top infill layer". This is different than 2.3.3 where it was sliced as "internal infill layer". This is undesirable because in a multi layer object where part of the first layer is a visible top surface it cannot be treated as a top layer and ironed. 
NOTE: In the attached image I have turned off ironing in the PrusaSlicer preview view to show difference in the single layer object on the left and the two layer object on the right. I have also attached the 2.4.0 beta3 .3mf file
[image]
[2.4.0 beta3 ironing test .zip](https://github.com/prusa3d/PrusaSlicer/files/7697754/2.4.0.beta3.ironing.test.zip)

bubnikv, 2021-12-12:
I see. 2.3.3 classifies a single layer (with zero bottom layers set) as internal infill, 2.4.0-beta4 as internal infill, however it is expected that the single layer will be classified as top infill in this case. Usually at least a single bottom layer is set, in which case the single layer is marked as bottom (bottom layers have precedence over top layers for single layer objects), then the single layer would be marked as bottom.

closed 2026-09-29 by prusaslicer-bot[bot]