prusa3d/PrusaSlicer #7090: Experimental purge tower has no first layer anymore

Moyrn, 2021-10-10:
### Version
2.4.0-Alpha3

### Operating system type + version
Wndows 10 20H2

### 3D printer brand / version + firmware version (if known)
Prusa MK3S-MMU2S + 3.10

### Behavior
Experimental purge tower has no first layer anymore to ease further layers adhesion
The skirt is done letting an empty space
First layer has special settings for proper bed adhesion
then when the first real purge occurs, for instance at Z 6mm, the  purge does not stick on the bed and stays around the nozzle
and quickly the print is ruined

PS: your painting feature is amazing ! i was testing it...

in 2.3 version, experimental purge tower performs the first layer on bed then "forgets" the purge until it is needed.
Please note that only problem is then at the end of the print, it purges at max Z in the air and not at latest used Z.

* _Steps needed to reproduce the problem_
paint an area at Z > 0, use at least 2 extruders with experimental purge tower
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
[BabyYodaFinal.zip](https://github.com/prusa3d/PrusaSlicer/files/7317002/BabyYodaFinal.zip)

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

SomeoneElse58, 2021-10-10:
The layer height scroll bar is more like a timeline when using that feature. If you scroll to 0.07(216) you will see that is when it will print the first layer.

I've 'successfully' used it in a print [2 days ago](https://www.twitch.tv/videos/1171234578) while having the same 'issue' as you of it not appearing in the slicer on the first layer.

Moyrn, 2021-10-10:
hello

in fact i printed the project, but the tower was really not there. only
some sparse "wipes" which did not take hold on the bed

thanks

SomeoneElse58, 2021-10-10:
> when the first real purge occurs, for instance at Z 6mm, the purge does not stick on the bed and stays around the nozzle and quickly the print is ruined

You may just have bed adhesion issues to deal with.

I loaded your 3mf file as a project into alpha3 and the only settings I changed were the two filament choices. It's going quite well and the wipe tower's [first layer](https://www.twitch.tv/videos/1173177002?t=0h50m30s) was printed as expected.

Moyrn, 2021-10-11:
> I looked at the gcode and don't see any issue. Adhesion problems are the
> easiest explanation. Please, let us know what you find out.

Hello

I will try to make another test.

I had double thoughts, and double checks on the matter.

...The change I see is that :
- in 2.3.1 the base of the purge tower is made at the beginning, possibly
with "first layer" settings (speed & height n - G1 F1200 like skirt  in the
gcode)
- while in 2.4.0-a3 it is made when required. In my project at layer 218 -
possibly with normal speed & height (G1 F3522 up to G1 F10608 in the gcode)

So my concern is not the "when", but the "how" the tower basement is done.

So yes, adhesion concerns, but not only.
I assume keeping the "first" tower layer with "first layer" settings is a
must. These settings are made for this very purpose

Wilfried

pipcorona, 2021-12-25:
I am having the same issue. It used to print the bottom layer of the sparse wipe tower on the first layer of the print. Now it does it on the first filament change. Since it is printing the first layer during wipe operations, it is not complete and does not adhere well.

Moyrn, 2021-12-25:
Yes same here, even with latest version. It is ok with perfect bed level,
but is nonetheless less "reliable" than previous way (force first layer).
In my humble opinion, the filament cost avoidance is not worth the
additional % of fail.

Or to rephrase wiping on the bed seems less effective/reliable than wiping
on a first level raft.

Although my last print went fine, there is an added level of cross fingering

Merry Christmas 3d friends !

pipcorona, 2022-02-05:
I checked Beta 2.4.1 and it does not resolve this issue.

Moyrn, 2022-02-05:
Because they do not consider it as a issue I guess.
While not systematically an issue, this weakens the feature. And I always
hunt for reliability.

Possibly enable it with a dropdown list
purge tower : full, experimental/raft, experimental/bed

While the latter is the one which annoys us, and the experimental/raft the
one we favour !

puckpuck, 2022-02-07:
I'm coming in here to say I can no longer use sparse wipe tower on a texture sheet.  I have redone bed leveling and even moved the wiper tower to the center of the bed, with my part on the edge, and the same issue. 

I want the sparse wipe tower to return to where it was previously, where the 1st layer is always printed with the rest of the print.

Moyrn, 2022-02-07:
aye ! i don't use textured pei, but that must effectively be worse on it !
thanks for posting. together we may force them to acknowledge this is a
step back

Merci !

Wilfried

lukasmatena, 2022-02-08:
Before you force us to do something, could you please explain in how the situation differs from any print with toolchanges on the first layer? The full first layer was never printed in that case (even with the "normal" wipe tower), because the space was needed to do the toolchanges. And it worked.

The actual issue in this case might the **height** of the first wipe tower layer. It is 0.07mm, apparently inherited from the layer where the toolchange was triggered, while it should respect first layer height. IMHO that's what needs a fix.

puckpuck, 2022-02-08:
@lukasmatena 

Say you have a print with a color/tool change at layer 10.

In PrusaSlicer 2.3, the first layer of the part, and the first layer of the wipe tower are printed together.  Layers 2-9 are printed on the part, then on layer 10, the tool change happens on the wipe tower, on what would now be considered the 2nd layer of the wipe tower.  Since there is already the first layer down on the printbed for the wipe tower, the adhesion of subsequent layers is great.

With PrusaSlicer 2.4, the first layer of the part is done, but nothing for the wipe tower.  It is not until layer 10 of the part, that the first layer of the wipe tower is printed on the print bed.  At this time, the printbed temp, nozzle temp, and nozzle height are not optimal for first layer print, and adhesion problems occur.

I would want to return to the functionality that was present in PrusaSlicer 2.3 and before for the sparse wipe tower, where the first layer of the wipe tower is printed when the first layer of the entire print is done.

lukasmatena, 2022-02-08:
@puckpuck All I'm saying is that the fact that the first toolchange is directly on the bed alone should not be a problem, because that is the way how it always is when the first toolchange is on the first layer, even with the usual, non-experimental wipe tower. I suspect that the actual problem is that the first layer of the wipe tower is too thin, which means the ramming is too fast when it tries to achieve the volumetric flow rate that is requested.

Yes, forcing the first wipe tower layer would fix it, but I'd like to be sure I understand the problem first. And although this thread is full of people saying "I want this, I want that", a single person has posted a 3MF and it suffers from the 'height too low' issue. Blindly forcing the first layer will not change anything when the toolchange is on the first layer, so if that case also does not work, I'd want to know.

pipcorona, 2022-02-08:
Greetings,

I have attached a .3mf file, gcode from 2.3.3 and code from 2.4.0. Most of the prints I do have colors changes after the first layer, which is why I noticed it. I've tried on the satin sheet and the smooth PEI. Since some of the unloading is happening on the sheet, I just can't get a consistent adhesion on the experimental purge tower. I'm getting consistent adhesion everyone else on the print. What usually happens is there are pieces sticking up from the tower where the unload happens, and then as the nozzle passes back over it scrapes and pulls up the filament off the bed. I did not have that issue in 2.3 with the first layer being printed first. I've had to resort to using full sized purge towers, when the first 8mm of my prints are usually all the same color and wouldn't actually need it.

Maybe the behavior of the first layer could be tied to the experimental feature? So if the experimental feature is enabled, it prints a raft as the first layer of the purge tower.

[PurgeExamples.zip](https://github.com/prusa3d/PrusaSlicer/files/8025658/PurgeExamples.zip)

pipcorona, 2022-02-09:
@lukasmatena I sliced the same 3mf using that build and it takes the behavior of the experimental slice tower back to what it was prior to 2.4.0. Just looking at the slice it doesn't look like anything obvious broke. I tried disabling the experimental feature and also flipping the model so color changes occurred on the first layer and the wipe tower seemed to operate as expected.

Thanks!

Moyrn, 2022-02-09:
@lukasmatena 
tested it OK. performs as expected (as it was in 2.3.x) 
thanks for taking the time to listen, to implement & to propose THAT FAST !!!

Layer1 in your fix
[image]

Same file in 2.4.0 - layer 1 as well
[image]

/kneel

I owe you a Pivo !

closed 2022-03-30 by lukasmatena

Moyrn, 2022-04-01:
> Thank you both for the testing. The fix is now in master for the 2.5
> release. Closing.

many thanks for your feedback & constant support