prusa3d/PrusaSlicer #7103: Bug: Bridge infill not connected to bridge outline

Noxmiles, 2021-10-12:
### Version
2.3.0 and 2.4.0-alpha3

### Operating system type + version
Windows 10 Pro 21H1

### 3D printer brand / version + firmware version (if known)
Prusa Mini+ firmware 4.3.2

### Behavior
* I want to print something that is partly free floating in the air. The bridge setting "Detect bridging perimeters" is active. The bridge outline is printed fine in the air, the bridge infill is not touching the outline and is falling down because of that.
* The infill bridge should overlap the outline bridge, so it's not falling down.

_Is this a new feature request?_ New Bug

#### Project File (.3MF) where problem occurs

[Kleiderhaken3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7333319/Kleiderhaken3mf.zip) (this file is from PrusaSlicer 2.3.0)

[image]

[image]

Noxmiles, 2021-10-19:
To better illustrate the problem here is a photo of the result. (Prusa Mini+, 0.2, only the affected section)

[image]

kubispe1, 2022-09-19:
It is not common but, I close this issue and let newer #7147 open. There are more comments. 
Thank @Noxmiles for report.

closed 2022-09-19 by kubispe1