prusa3d/PrusaSlicer #7109: Slicer creates random zits on MMU painted model

baz4, 2021-10-13:
### Version
2.4.0-alpha3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
When slicing an MMU painted model, sometimes the slicer produces random 'zits' on the model:

[image]

The zits appear on one layer only, are always the default colour and seem to be outside of the model boundary:

[image]

Could this be similar to #7104 but the dimple goes out of the model instead of into it? and as there is no actual geometry, the default colour is used?

#### Project File (.3MF) where problem occurs
This appears to happen on more complex models. Unfortunately, all the the models I have seen it on are not free models, so I can't upload here. I will upload an example if I can replicate with a free model.

hejllukas, 2021-11-05:
Thank you for reporting this issue. I was able to reproduce this issue on one model, and it should be fixed with e13be902836e52b0f859be00dbe39fd96d5b035a. Could you please verify that this [build](https://drive.google.com/file/d/1YaVtgg-ySB5X2m8tdwZODIXFJPVQG7Vr/view?usp=sharing) (or you can compile the current master branch) also fixes this issue on your models?

baz4, 2021-11-08:
Thanks for fixing this issue. I've tested the latest master branch with a couple of models that had the problem and the zits are no longer there. The models are looking really clean now, thank you.

hejllukas, 2021-11-08:
Nice! Thank you very much for your help with testing.

closed 2021-11-08 by hejllukas