prusa3d/PrusaSlicer #7112: Default colour appearing between two MMU painted areas

baz4, 2021-10-14:
### Version
2.4.0-alpha3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
When painting a model, sometimes a small area of default colour appears between two painted areas once sliced. The default colour section does not appear on the model and cannot be painted.

Example of a painted model:

[image]

Once sliced, the model looks like this:

[image]

The blue section should not be there. It doesn't appear on that seam further down or up the model. It always occurs when there is an angle between 2 colours, I don't think I have ever seen this happen on a flat surface with multiple colours. It is possible that the problem is related to the size of the angle between the two colours, but I have not been able to prove this theory yet.

#### Project File (.3MF) where problem occurs
[mmu-painting-default-colour-showing.zip](https://github.com/prusa3d/PrusaSlicer/files/7345898/mmu-painting-default-colour-showing.zip)

jb221, 2021-11-01:
I'm super eager for this to get fixed, I've got 10 heavily-multicolor projects waiting to print because every one of them has this issue. I hope my printer is enjoying its rest because it's gonna be working non stop when they get this sorted 🤣

bubnikv, 2021-11-01:
Did you test with beta1?

jb221, 2021-11-01:
> Did you test with beta1?

Checked it last night with beta1 and issue is still present. Granted I used the .3mf files that I painted using alpha3 but the issue was still present. When I get back home I can try repainting in beta1 to see if that produces different results.

baz4, 2021-11-01:
> 
> 
> Did you test with beta1?

I've just tested with beta 1, painting the model again after opening the STL rather than the above 3mf project. The problem is still there.

hejllukas, 2021-11-02:
Thank you for reporting this issue. It was fixed with 7ce13b4a26e69c96296b801d363d0da65a623674, and it will be part of the next beta release.

closed 2021-11-02 by hejllukas