prusa3d/PrusaSlicer #7114: "Ask for unsaved changes when creating new project" Option not working

geoff0616, 2021-10-14:
### Version
2.4.0-alpha3+win64
### Operating system type + version
Windows 10
### 3D printer brand / version + firmware version (if known)
Prusa MK3S
### Behavior
New "Ask for unsaved changes when creating new project" option which, if enabled, prompts a window after confirming if you'd want to begin a New Project. But issue is when this option is disabled, no changes are done and window still prompts when selecting "New Project" or executing the Ctrl + N shortcut. 

Old prompt was able to be dismissed via keyboard controls (i.e. hitting Enter), new prompt requires Alt+Y Alt+N selectors.

_Is this a new feature request?_
No, but possible new feature request to be able to dismiss prompt with single keyboard input.

david-mikulin, 2021-11-11:
Verified that this still exists on 2.4.0 Beta 1. Tested on Apple MacBook Pro w/ M1 Pro chip, Mac OS Monterey 12.0.1.

Steps to Reproduce:
1. Started Prusa slicer
2. Turned off the "Ask for unsaved changes when creating new project" option
3. Verified that New Project menu option was greyed out.
4. Imported a STL file, but made no changes.
5. Selected "New Project" and window prompting for confirmation appeared.

rtyr, 2021-11-11:
Reproduced on master. @enricoturri1966

david-mikulin, 2021-11-11:
Also, noticed that the same confirmation window appears if you try to quit the application after importing an STL, but make no changes to it.

bubnikv, 2021-12-10:
The issue here is the wording of "Ask for unsaved changes when creating new project". The text is inherited from PrusaSlicer 2.3.3, where only changes in presets were checked for. Now we check for changes in the project itself, namely the plater. 

We shall reword "Ask for unsaved changes when creating new project" to somethink like

"Ask for unsaved changes in presets when creating new project"

Ali229, 2021-12-20:
This issue is also occuring while closing the application.
Steps:
1. Uncheck "Ask to save unsaved changes when closing the application...".
2. Make a change and close the application.
3. Prompt comes to save the project when it shouldn't.

bubnikv, 2021-12-21:
I believe this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7114
namely, see my comment https://github.com/prusa3d/PrusaSlicer/issues/7114#issuecomment-991132153

> Old prompt was able to be dismissed via keyboard controls (i.e. hitting Enter), new prompt requires Alt+Y Alt+N selectors.

This is a different topic we already handled at the "Send to OctoPrint" dialog recently.

Ali229, 2021-12-21:
What do you mean? This is #7114.

Ali229, 2022-01-14:
Any update on this? Annoying bug!

lukasmatena, 2022-01-14:
@Ali229 Yes. I have just written about this when writing a change log for the next release:
>  A new option "Ask for unsaved changes in project" was added into Preferences. The original three options "Ask for unsaved changes when..." always refered to changes in **presets** and were now reworded so it is clear and does not lead to confusion #7114

bubnikv, 2022-01-27:
It will be fixed in the upcoming PrusaSlicer 2.4.1-beta1.

closed 2022-01-27 by bubnikv