prusa3d/PrusaSlicer #7123: /* Feature Request */ Angled supports

ephramcohen, 2021-10-16:
### Version
V2.3.1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2s

### Behavior
I have a model that is on a base.  The middle of the model needs supports.  I do not want the supports going straight down from the model to the base.  

The other place where this occurs is where the model itself is directly below the area that needs support

Instead of placing supports on the top face of a model, I want the supports to go at an angle from the model, to the print surface, away from the base.  This way the support does not leave any marks on the base itself which is a very visible place.  There will be marks on the model, but seeing as this is the underside of the model, it is in a much less visible place.

I would be OK with a "paint it" type approach, where I could mark (or have prusaslicer mark) the area that needs support, then I could choose a "landing zone" for the support to attach to the build plate.

In the uploaded zip file, I include a model that has the problem, and an stl that I modified that illustrates one possible solution.

_Is this a new feature request?_
yes

#### Project File (.3MF) where problem occurs
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/7357920/files.zip)
_

rtyr, 2021-10-16:
Isn't that a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1514?

ephramcohen, 2021-10-16:
Related, yes, but I would say not a duplicate.

The key important difference is there is no branching.  This is just the normal support that Prusaslicer has, just angle it in the Z dimension so that it "misses" the rest of the model and lands on the build plate next to the model.

ephramcohen, 2021-10-16:
Here is an openscad example that I think illustrates what I am asking for.  The traditional solution is in red, my proposal is in green.

// a sphere with 4 supports and a base
translate([0,0,20])sphere (10);
cylinder(1,10,10); // base
translate([7.5,0,0])cylinder (15,1,1);
translate([-7.5,0,0])cylinder (15,1,1);
translate([0,7.5,0])cylinder (15,1,1);
translate([0,-7.5,0])cylinder (15,1,1);

//traditional
color("red")cylinder (10.5,1,1);

CubePoints = [
  [  8,  8,  0 ],  //0
  [ 9,  8,  0 ],  //1
  [ 9,  9,  0 ],  //2
  [  8,  9,  0 ],  //3
  [  -0.5,  -0.5,  10.5 ],  //4
  [ 0.5,  -0.5,  10.5 ],  //5
  [ 0.5,  0.5,  10.5 ],  //6
  [  -0.5,  0.5,  10.5 ]]; //7
  
CubeFaces = [
  [0,1,2,3],  // bottom
  [4,5,1,0],  // front
  [7,6,5,4],  // top
  [5,6,2,1],  // right
  [6,7,3,2],  // back
  [7,4,0,3]]; // left

//proposed
color("green")polyhedron( CubePoints, CubeFaces );

ephramcohen, 2021-11-02:
Thank you bubnikv for the image.  The coloration of the two support styles got lost, so just to be 100% clear.  

The current style is the middle upright that stands on top of the base.  The feature request is the angled component that misses the base.  Perhaps 2 supports are needed for this particular model.

I was thinking that with the new paint-your-support interface, the "landing zone" could be painted also and it's a linear transform to calculate the position of the support at any height.

closed 2026-09-29 by prusaslicer-bot[bot]