prusa3d/PrusaSlicer #7132: [Feature Request] Align Selected Models

Alpacad, 2021-10-18:
### Version
PrusaSlicer-2.4.0-alpha3+win64

### Operating system type + version
Windows 10 Pro 21H1 / Windows 11 Home 21H2

### Description
Add option to align selected models, for example drag and drop five models from a folder to PrusaSlicer, then right-click on one of them while they're all selected (or first select the ones you need to align) and click on "Align Selected Models".

### Notes
I know there's already an option called "Add part" but this would require to drop a single model, then click the editing icon near the models list, go to Add part>Load and then navigate to the models to be aligned and select them. This way could be more fast and simple.

### Comparison to other slicers
There's already this feature in ideaMaker.

_Is this a new feature request?_
Yes

Ali229, 2021-12-20:
Need this feature!

FidelCapo, 2022-02-22:
Alignment of selected models can be done with the shortcut shift+A. Is this what you are looking for?

Alpacad, 2022-02-22:
That's not it, I've tried in case I missed some functionality of that shortcut, but no.
It works for STL, which do not store positioning data, but 3MF files do store position and it can be used to print multiple objects in place. An example could be inlay 3D printing.