prusa3d/PrusaSlicer #7136: Custom extrusion width for first layer width does not effect raft.

Wallby, 2021-10-18:
### Version
2.3.3+linux-x64

### Operating system type + version
LMDE4 Debbie (Cinnamon 5.0.5, linux 4.19.0-17-amd64)

### 3D printer brand / version + firmware version (if known)
Not relevant (custom printer, not a preset).

### Behavior
Settings.. Extrusion width for first layer is 1mm, default extrusion width is 0.45mm
* with 1 raft layer, the first layer looks like this..
[image]
* with 0 raft layers, the first layer looks like this.. 
[image]

The reason that the first layer is 1mm for me, is because when I set it "too thin", the print won't stick (using glass). The 1mm thick lines print fine with the raft, but the "raft lines" don't stick.

#### Project File (.3MF) where problem occurs
Ask if desired.

Wallby, 2021-12-08:
@bubnikv tested (on windows 10 this time for convenience, but still build 2.4.0-beta3).. figured out that the extrusion width for the raft is the "support" extrusion width. This works in 2.3.3 as well.
The issue persists.. i.e. in the following image there are two raft layers and for the light green raft layer the first layer width is used..
[image]
In the following image, however, there is only one raft layer (the dark green one) to which the first layer width is not applied (though it is the first layer).
[image]

Wallby, 2021-12-08:
It seems that knowing this it is possible to just add two raft layers instead of one and that would be a solution, but this still seems a bit weird no?

bubnikv, 2021-12-08:
thanks for investigation

closed 2026-09-29 by prusaslicer-bot[bot]