prusa3d/PrusaSlicer #7137: [BUG] terrible underextrusion on layer start after aligned seam

neoxm3, 2021-10-18:
### Version
All version from 2.1.1 (not included) to 2.4.0 alpha3

### 3D printer brand / version + firmware version (if known)
Kingroom Kp3s + Marlin 2.0.9.2 . Machine calibrated, drivers in uart mode, 16microsteps, titan extruder.

### Behavior
* The problem: 
This problem has been commented on several times and I think it is a "serious" problem, because it is still there, it is not corrected and in the end the threads of the people who point to the problem always end up closed due to inactivity since as it is not provided solution, people get a little desperate ...

The problem is very simple since whatever you do and configure what you configure, in the versions of prusa slicer cited, after the seam and at the beginning of the next layer (it occurs in this particular case on vertical walls without filling , walls that are thin and perimeters are enough to close them) a lack of extrusion occur, terribly affecting the final appearance.

The funny thing is that in many threads it is pointed out that it is due to the use of the aggressive cooling routines, but I have tried to disable it myself, and the problem is exactly the same.

* Steps needed to reproduce the problem:
Here is a file, with the project for the latest version of prusa slicer, with all parameter configured. Printed with PLA, esun 1.75 black silk, very good quality.
https://drive.google.com/file/d/1IhyRv09KBFRHhydswp6mG9fBvef8TsHd/view?usp=sharing

* Expected Results:
This are the espected results, and the expected results are usin the same file, in prusa slicer 2.1.1., the two prusa slicers have diferent parameters (some clearly as they are not in the old version)  but you just open it, accept the import and we prepare the file, not changing anythin.
https://i.ibb.co/kmNmzLv/foto1.jpg

I have uploaded a video where the finish can be seen better, what to say, which is incredible ... Despite the fact that in this version of prusa slicer, due to its age, we cannot put the seam where we want and it is not perfectly aligned
https://drive.google.com/file/d/1yOzxtbq8FOfzjuojqAH_sqqDCOC4OBsL/view?usp=sharing

* Actual Results:
From prusa slicer 2.1.1 (not included) to 2.4.0 alpha3, the incredible result of the layers it's terribly bad after the seam...
https://i.ibb.co/VxBmfKD/foto2.jpg

Uploaded video:
https://drive.google.com/file/d/1TNyl0fw0jSHNP3HOyUigcyvOZZ_aeYBb/view?usp=sharing

And i upload a video to see the diference with these two pieces:
https://i.ibb.co/YjTPk8w/foto3.jpg
video:
https://drive.google.com/file/d/1oFnkjDJe23vY8TR9m2dFXlpfKQTidO20/view?usp=sharing

for me, this quality is not valid.
------------------------------------------------------------------------------------------------------------------------------------

The difference is clearly seen, it is some fault that has remained since the next update from version 2.1.1, and I have even come to think that it may be a problem that for whatever reason occurs only on machines that are not original prusa and yes in other of other companies?

It is very rare, but I think it should be prioritized because you can see in the video, prusa slicer is incredible, it has amazing functions and leaves a very good finish, but details like this totally ruin a piece and it is something that can be reviewed on the forum, it's happening to a lot of people ...  

And finally, it is not the result of chance, those same pieces have been printed (the ones that look bad) 10 times, always being exactly the same, changing different for meters (I have tried with and without retraction, with and without extra lenght on restart, with and without layer fan, with and without cooling routines ... and the result is always the same.
In prusa slicer 2.1.1, I only printed one ... and the first time, perfect.

I hope a solution appears, thanks for reading everything!

dnkr0z, 2021-12-10:
Hello. Using Anycubic I3 Mega S with DualDrive Direct Extruder based on e3d V6 cloned parts. Printer powered by klipper firmware v0.10.0-167. 
I have similar problem with my printer, and i can't solve it too. I tryed to reproduce your experience with PS 2.4.0 beta3 and PS 2.1.0 relase versions
On photo you can see results of this experiment
https://drive.google.com/file/d/16CxqbCIEyXNjV1VE1Gc1j5769Ev0FwxA/view?usp=sharing

dnkr0z, 2021-12-13:
Try to disable "external_perimeters_first". And "detect_thin_walls" for this model.

closed 2026-09-29 by prusaslicer-bot[bot]