prusa3d/PrusaSlicer #7146: Reset settings for spiral vase mode is inconsistent

mroek, 2021-10-21:
### Version
Version 2.4.0-alpha3+win64

Steps:
1) Enable spiral vase mode from a normal settings profile
2) Click OK in the dialog box that asks if Prusaslicer should change other settings in order to use spiral vase mode
3) Click the back arrow in the top menu line, which should reset all settings in that option group, including spiral vase

Instead of just resetting the settings, the dialog box from 2) above pops back up again, which I think it shouldn't do, as the reset command should turn off spiral vase mode. It is simple enough to just answer "No" in this dialog box to get the expected result, but the dialog box should not have popped up again. It is simply confusing.

This is just a very minor bug, and one that is easy to work around, but you might want to change this behaviour.

bubnikv, 2021-12-11:
Fixed in the upcoming PrusaSlicer 2.4.0-beta4.

closed 2021-12-11 by bubnikv

mroek, 2021-12-11:
Great, thanks.