prusa3d/PrusaSlicer #7147: Bridging direction is poorly chosen

yoyo42, 2021-10-21:
### Version
Version 2.4.0-alpha3+win64

### Operating system type + version
Win 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s

### Behavior

When sliced the 0.4mm version of the calibration cube https://www.prusaprinters.org/prints/69934-helix-test-cube-filament-sample-xyz-calibration has two identical runs of bridging but on different axes, above the 'X' and 'Y letters. On the X side the bridging is perfect, on the Y side it does two lengthwise runs either side then tries to bridge sideways between them which fails with lots of loops hanging underneath. The bridging should run between the corner pillars in both directions.

[image]

#### Project File (.3MF) where problem occurs

[04HelixTestCube.zip](https://github.com/prusa3d/PrusaSlicer/files/7390626/04HelixTestCube.zip)

idlebear, 2022-02-27:
Hi there - I just came across this problem when slicing this [model](https://www.prusaprinters.org/prints/140928-bridges-of-cubison-county). Resulted in some of the bridges running in the correct orientation (across the gap) and some perpendicular to it.  As a result, the bridges sag for the first layer.  Here's the slicing result from win 2.4.0:

[image]

The same file, sliced in the current SuperSlicer release:
[image]

Thanks

yoyo42, 2022-03-18:
Thanks, your second image is exactly the way I think it should be done.

Noxmiles, 2022-12-12:
Nothing new? The first report of this issue is from october 2021. So, a bump for this.

hagre, 2023-03-07:
It looks like it is still existing in 2.6.0-alpha5
[image]

Raynoodles, 2023-03-25:
Can confirm with hagre with the same file yoyo24 used in 2021.
[image]

AinoSpring, 2024-01-05:
Sadly still exists in 2.7.1.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]