prusa3d/PrusaSlicer #7198: Auto Slice missing after Configuration Update

dartrax, 2021-10-29:
### Version
2.4.0 beta 1

### Operating system type + version
Win 10

### Behavior
* _Describe the problem_
While viewing the preview, I noticed a "Configuration Update" notification and clicked on "See more". A Message box asked me if the configuration update should be applied. I confirmed. After the configuration update, the Preview was gone.
* _Steps needed to reproduce the problem_
  * ensure having background slicing enabled
  * Slice a model and preview
  * Click on "See more" in a config update notification at the lower right corner of the preview window
  * Confirm the update
* _Expected Results_
Auto reslice after config update if background slicing is enabled
* _Actual Results_
Preview is cleared after config update

_Is this a new feature request?_
no

bubnikv, 2021-11-01:
That is a valid point, a small inconvenience. @YuSanka will take care of it one day.

dartrax, 2021-12-04:
Seems to be fixed in 2.4.0 beta 3? Resliced correctly today, however, tested only once.

bubnikv, 2021-12-08:
If it is fixed, let's close it :-)

closed 2021-12-08 by bubnikv