prusa3d/PrusaSlicer #7200: Allow different extrusion multipliers for different features, layers, modifiers, etc

shadowphile, 2021-10-30:
2.3.1+Win64

_Is this a new feature request?_
Yes,  I would like to deliberately under or over extrude in different operations and places.  Can only do global right now. 
(some have suggested in the forums that changing the extrude width is the same thing, but it's not)

ane109, 2021-12-14:
I second this request. As an example, this would be extremely useful for light/optical diffusers. If the extrusion multiplier is too low, the diffuser will suffer from uniformity issues, however the extrusion multiplier value that is ideal for uniform diffusion leads to excess buildup of plastic on the nozzle that can ruin prints. A feature like this would allow for a more uniform diffuser with less plastic buildup on the nozzle over the course of the print.

wooddragon48, 2022-03-18:
I definitely second/third this request, as it wood allow me to fine tune my prints to be much nicer. I specifically would like to fine tune the extrusion multiplier for outer perimeters and top infills. Higher on the former and lower on the latter, as currently I am using a multiplier that is in the middle but not ideal for either.

chschmit, 2022-06-15:
I upvote this request. 
I've found that specific colors are' denser' even within the same brand of filament. The extruder doesn't bite into them as much and changes the flow rates when I change colors. I want the ability to change the flow rate/multiplier for each filament/color change.

schleifi, 2022-12-15:
Would be great to have different extrusion multiplier for inner and outer perimeter.
My use case: outer perimeter with normal settings, guaranteeing that the object keeps its accurate dimensions, inner perimeters increased E.M. to solidly fill also inner corners and bring in stability.

zenturacp, 2023-03-25:
It would be highly appriciated if you could adjust the flow for the toplayer - of all top solid infill surfaces. not just the top top layer.

It's indeed very easy to tweak print quality with the ability to modify the top layer extrusion multiplier, if you want to dial it in perfectly

TomFrost, 2023-04-04:
I've recently switched from Cura to Prusaslicer and this is the only thing I miss. The ability to set an extrusion multiplier for exterior perimeters vs interior vs infill, bottom vs top, even for the different perimeter types on the bottom, allows me to fine-tune my dimensional accuracy as well as cure elephant's foot in a much more consistent way than a simple first layer scaling operation. I can easily achieve the same dimensional accuracy for inside-to-outside perimeter drawing as outside-to-inside, which is incredibly difficult to do when extrusion width is the only available control.

Chris-Jayden, 2023-05-25:
I'm really hoping they still plan to add independent control for different feature types with respect to both Extrusion Multiplier and Linear/Pressure Advance. This very much old request needs much more attention.

iqubik, 2023-07-30:
как странно что этого до сих пор не сделали

ugryy, 2023-08-04:
The only reason I stop using prusaslicer everytime. My step and flow calibrations are perfect, yet with 0.4mm top layer width I got very bad results. Tried 0.32 and even 0.28, they are close to very smooth top layer but then on most of the models there are gaps on top layer where it connects to walls... 
With superslicer and orcaslicer, %80 top layer flow and 0.4 width; tadaaa you have perfect top layer everytime.

Just please add the damn thing 😞

JPK-MG, 2023-09-07:
Same here... It would be nice to have flow rate options for top layers.

I was recommended PrusaSlicer over Cura and I'm really happy with the experience, so far. I'm using Silk filaments for multiple prints of the same model and everything hinges on my ability to get a nicely extruded top layer (without ironing). It's impossible to get a top layer that isn't slightly underextruded, leaving unfilled tracer lines between sections of the infill, which would be rectified by setting top layer flow to 105-110%. Unfortunately, I have to print this model in Cura to get that perfect top surface.

My appreciation goes to anyone who implements this feature.

Zomflax, 2023-10-23:
I am also waiting for this feature to be added... This is so helpful in other slicers like Cura. 110% for everything inside and 100% for outer perimeters is my usual settings. This gives me the strongest parts and keep them very nice and smouth.

Please add this option.

Thanks!

Eastcoastcomposites, 2023-12-10:
Yes we definitely need a first layer extrusion multiplier/ flow adjustment. Im not sure why there hasnt been one added by now. Ive been using prusaslicer for about 3 years figuring it would get added. I have to manually adjust the flow every single print on the printer interface and its very cumbersome and annoying.

rogery555, 2023-12-15:
I cant believe its not possible to adjust the flow in the height range modified setting????

cedlap, 2023-12-23:
It's annoying, but in the AFTER_LAYER_CHANGE custom gcode, use this. At least you won't have to mess with gcodes everytime.

{if layer_num==0}M221 S150{endif} #set the flow to 150% or whatever
{if layer_num==1}M221 S100{endif} #set the flow back to 100%

We need actual support for this feature.

arshish1612, 2024-01-12:
Wonder why no one has picked this up for dev... It's such a useful feature that is omitted

TheShanMan, 2024-03-01:
Please at least start with a first layer multiplier option. I'm trying to transition over from cura and not having this very useful setting is disappointing, but seeing that this request has been languishing for a long time is discouraging.

arshish1612, 2024-03-09:
This is how I manage it:

[image]

ProfDrYoMan, 2024-03-09:
Since I needed a feature like this for a calibration I spent some time to code a pre-processor. 

https://github.com/ProfDrYoMan/prusaslicer_m221_per_object

Maybe it helps somebody.

lutorm, 2024-05-08:
Another use for this would be to increase the extrusion multiplier for supports. Currently this is controlled by the support extrusion width setting, but that is a printer setting and not a filament one. I find that I need a higher extrusion width for PETG supports for them to adhere properly, but now the only way to edit the setting is by changing the printer preset.

gopgnz, 2024-06-11:
PLEASE add the ability for us to change the extrusion multiplier for each feature. I need a convenient way to adjust the flow rate for the most top surface in particular. Even when the flow rate is calibrated nicely for each filament the most top surface is often underextruded for me on my Prusa XL when printing larger objects (even with 5-6 top layers). Using the g-code substitution method is way too cumbersome.

arshish1612, 2024-07-21:
Awaiting an update

breathless19, 2025-02-09:
This is a much needed feature!

suxscribe, 2025-03-21:
+1. Please 🙏

cheatreal321, 2025-04-11:
Have to add to this and say this is a sorely missing feature. After much tuning my Prusa XL prints nearly perfectly except for the top solid infill which is consistently underextruded regardless of filament. I can't increase the extrusion multiplier or flow rate any further otherwise my nozzle completely gunks up.

arshish1612, 2025-07-27:
+1 Need to keep this request alive. Lets not let this go stale

digtechx, 2025-12-18:
+1
The extrusion multiplier is set to the same value for the entire print without the possibility of changing it during printing. Binary G-code cannot be effectively edited and the use of G-code is also very complicated.

The function should be able to be inserted into any layer.

This setting should be good for the following options: increasing strength, solving the first layer of the print on supports (e.g. Prusa-Mini), better fine-tuning the top layer e.g. Prusament PC-Carbon Fiber.

Siegsaw, 2026-04-07:
+1
This would definitely help, especially when printing transparent filaments. I print with clear pctg and i have to use a global of multiplier of 1.05 for clarity, but it leaves the outer walls far too thick and reduces print accuracy.

Tupson444, 2026-09-12:
still relevant