prusa3d/PrusaSlicer #7201: Sidewinder X1 BL-TOUCH update K factor

plerros, 2021-10-30:
### Version
2.3.3

### Operating system type + version
Fedora 34

### 3D printer brand / version + firmware version (if known)
Artillery Sidewinder X1 with bl touch
Marlin 2.0.9.1 (modified)

### Behavior
In the profile Sidewinder X1 BL-TOUCH, the start G-code does:
```
M900 K0.12 ; K factor
```
That value is too low for a volcano hotend.
The marlin config profile has a k factor of 0.22.

rtyr, 2021-10-30:
@SzabolcsHornyak is author of the profiles.

Somehow I doubt that K0.22 is correct value for direct drive. It also depends on material type. It's always safer to use lower value than excess value.

Is the value K0.22 what you get with Marlin test pattern?

plerros, 2021-10-30:
I haven't printed the test pattern.
From my experience 0.22 works fine.
I have noticed that 0.12 leaves small gaps in corners.
I'll print the test pattern and report back.

rtyr, 2021-10-30:
> 
> 
> 0.12 leaves small gaps in corners. 

That is exactly what happens if your value is too high, not too low.

plerros, 2021-10-30:
[image]
0.14 looks good

plerros, 2021-10-30:
I re ran the test twice.
[image]
[image]
Here 0.18 looks good.

rtyr, 2021-10-30:
So K0.12 seems like a good default value. It should be conservative, because it will be different for each material (for example higher for PETG, lower for PLA) and also the temperature plays a role. Lower value is safe, it doesn't cause issues.

It is better to put M900 command to the filament settings instead (custom gcode section). But I get why author of the profiles didn't do it, because filament profiles are shared for more Artillery printers.

I think we can close this issue. K0.12 seems to be good default value.

closed 2021-10-30 by plerros