prusa3d/PrusaSlicer #7202: New Feature Request: Sequential Printing of Parts of an Object

zzjlamb, 2021-10-30:
Currently, objects can be printed sequentially, however the slicer does not allow parts of an object to be printed sequentially, because they are by definition too close to each other so a collision with the print head seems likely.
Other slicers, for instance Simplify 3D, do allow parts of an object to be printed sequentially. 
This is useful in situations where elevated features of an object, for example narrow towers on a flat base, are widely separated horizontally. It is hard to prevent stringing in this situation, particularly with stringing prone filaments like PETG or TPU.
The height of such features would be limited by the topology of the print head, and height of the lowest X axis rod, however even if the towers were are few centimeters apart then about 3mm of layers (30 layers at 0.01mm layer height) could be printed without any travel between features, and if the features were more than say 40mm apart, then the height of the features above the base could be up to around 70cm or so, or even much more if they were close on the Y axis and narrow.
This would also save some print time by reducing travel, but mostly would improve print quality.

bubnikv, 2021-12-08:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/7202

bubnikv, 2021-12-08:
and https://github.com/prusa3d/PrusaSlicer/issues/3600

bubnikv, 2021-12-08:
and https://github.com/prusa3d/PrusaSlicer/issues/828

dzwiedziu-nkg, 2026-09-04:
I'm realized it by plugin for Prusa Slicer 3.0: https://forum.prusa3d.com/forum/prusaslicer/thank-you-for-the-plugin-direction-in-3-0-three-slicing-plugins-it-made-possible-and-a-pr/

Working in my fork because main branch not implemented hooks for my plugins yet.