prusa3d/PrusaSlicer #7204: 2.4.0 Beta 1 crashes when uploading to Octoprint

roffe, 2021-10-30:
### Version
Version 2.4.0-beta1+win64

### Operating system type + version
PrusaSlicer
Version:   2.4.0-beta1+win64
Build:     PrusaSlicer-2.4.0-beta1+win64-202110251333

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22000), 64-bit edition
Total RAM size [MB]: 34,196MB
OpenGL installation
GL version:   4.6.0 NVIDIA 472.12
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2080/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 660 lines cut ...]
OsInfo[31].Value=Professional
OsInfo[32].Key=ring
OsInfo[32].Value=Retail
OsInfo[33].Key=expid
OsInfo[33].Value=NG:151E91,DR:795E2,DR:7DAC3,DR:91BE1,DR:957E7,DR:9729B,DR:9CE15,FX:114145DF,FX:1201898A,FX:120394A5,FX:1203F7EE,FX:1204F6A4,FX:12050417,FX:12050830,FX:12050869,FX:1211834E,FX:12118457,FX:12118488,FX:12169303,FX:1216A286,FX:1219B35D,DR:9D64E
OsInfo[34].Key=fconid
OsInfo[34].Value=18299130,0,2,0;19638787,0,2,0;23531064,2,2,0;23562335,2,2,0;23563673,2,2,0;25704915,1,2,1
OsInfo[35].Key=containerid
OsInfo[36].Key=containertype
OsInfo[37].Key=edu
OsInfo[37].Value=0
OsInfo[38].Key=servicinginprogress
OsInfo[38].Value=0
FriendlyEventName=Stopped working
ConsentKey=APPCRASH
AppName=PrusaSlicer
AppPath=C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=23EF93122A07220D9F4508A12D3E257C
MetadataHash=-1685861307
```

#### Project File (.3MF) where problem occurs
Any file, does not matter

roffe, 2021-10-30:
```
PS C:\Users\flejd\PrusaSlicer> .\build_win.bat -d=..\PrusaSlicer-deps -r=console -c=Debug
File not found - .DEPS_PATH.txt
0 File(s) copied
File not found - .DEPS_PATH.txt
0 File(s) copied
**********************************************************************
** Build Config: Debug
** Target Arch:  x64
** Build Steps:  app
** Run App:      console
** Deps path:    C:\Users\flejd\PrusaSlicer-deps
** Using Microsoft Visual Studio installation found at:
**  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.9.4
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
Cleaning build ...
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22000.
-- The C compiler identification is MSVC 19.28.29914.0
-- The CXX compiler identification is MSVC 19.28.29914.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: C:\Users\flejd\PrusaSlicer-deps\usr\local (from cache or command line)
Building with Win10 Netfabb STL fixing service support
-- Found Boost: C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.75.0/BoostConfig.cmake (found suitable version "1.75.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams 
-- Boost::boost exists
-- Found TBB: C:/Users/flejd/PrusaSlicer-deps/usr/local/include (found version "2017.0")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = C:/Users/flejd/PrusaSlicer-deps/usr/local/include
--   TBB_DEFINITIONS         = __TBB_NO_IMPLICIT_LINKAGE
--   TBB_LIBRARIES           = C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/tbb_static.lib
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/tbb_static_debug.lib
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/tbb_static.lib
-- Found CURL: C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/libcurl.lib (found version "7.75.0-DEV") 
-- Found EXPAT: C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/expat.lib (found version "2.2.0") 
-- Found ZLIB: C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/zlib.lib (found version "1.2.11") 
-- Found PNG: optimized;C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/libpng16_static.lib;debug;C:/Users/flejd/PrusaSlicer-deps/usr/local/lib/libpng16_staticd.lib (found version "1.6.35") 
-- Found OpenGL: opengl32   
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: C:/Users/flejd/PrusaSlicer-deps/usr/local/include
-- FindGLEW: GLEW_INCLUDE_DIRS: C:/Users/flejd/PrusaSlicer-deps/usr/local/include
[... 118 lines cut ...]
C:\Users\flejd\PrusaSlicer-deps\usr\local\include\boost-1_75\boost/unordered/detail/implementation.hpp(1452,7): warning C4996: 'std::allocator<T>::is_always_equal': warning STL4010: Various members of std::allocator are deprecated in C++17. Use std::allocator_traits instead of accessing 
these members directly. You can define _SILENCE_CXX17_OLD_ALLOCATOR_MEMBERS_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\flejd\PrusaSlicer\build\src\libslic3r\libslic3r_cgal.vcxproj]
C:\Users\flejd\PrusaSlicer-deps\usr\local\include\boost-1_75\boost/unordered/detail/implementation.hpp(1452,7): warning C4996: 'std::allocator<T>::is_always_equal': warning STL4010: Various members of std::allocator are deprecated in C++17. Use std::allocator_traits instead of accessing 
these members directly. You can define _SILENCE_CXX17_OLD_ALLOCATOR_MEMBERS_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\flejd\PrusaSlicer\build\src\libslic3r\libslic3r_cgal.vcxproj]
C:\Users\flejd\PrusaSlicer-deps\usr\local\include\boost-1_75\boost/unordered/detail/implementation.hpp(1452,7): warning C4996: 'std::allocator<std::pair<const K,T>>::is_always_equal': warning STL4010: Various members of std::allocator are deprecated in C++17. Use std::allocator_traits in 
stead of accessing these members directly. You can define _SILENCE_CXX17_OLD_ALLOCATOR_MEMBERS_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\flejd\PrusaSlicer\build\src\libslic3r\libslic [... cut ...]
Unbalanced parenthesis.

Total Build Time Elapsed 00:00:00,00

Running console application...
[2021-10-30 17:23:57.811444] [0x00006f84] [trace]   Initializing StaticPrintConfigs
[2021-10-30 17:42:59.376651] [0x00006764] [error]   OctoPrint: Error uploading file: , HTTP 409, body: `{"error":"Trying to overwrite file that is currently being printed"}
`

ERROR: *** Build process failed at run step. ***
```

it crashed directly on this.

is it maybe the error handling of the upload that is the problem?

roffe, 2021-10-30:
ok, no. i tried the same again, then it showd a dialogue with the error, tried to upload again, then it crashed without anything in the console

lukasmatena, 2021-10-30:
This is probably a duplicate of #7170, which is already fixed.

roffe, 2021-10-30:
ok then it's not fixed by #7170, i just built latest master and it's still there

roffe, 2021-10-30:
and mine is not every second, I can sometimes upload and generate multiple gcode before it crashes

roffe, 2021-10-30:
My bad, it's not merged, gonna try a build with it 👍🏻

roffe, 2021-10-30:
Confirmed, looking forward to Beta 2 release, will just roll my own built version until then.

Thank you!

closed 2021-10-30 by roffe