prusa3d/PrusaSlicer #7206: [Feature request] Print preview should use all CPU cores.

vladimir-aubrecht, 2021-10-30:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10 Pro x64 (Version 20H2, OS Build 19042.1288)

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
- Open any model in Prusa Slicer (ideally very complex one as it's easier to see the problem).
- Open task manager and switch it to visualize every core separately.
- Enable supports and switch to Preview view.
- Expected: Supports and estimated time should be calculated by using all cores (at least for complex models).
- Actual: Supports are indeed loading all cores when calculating, but estimated time is loading just 2 cores which leads to very long waiting time when waiting for estimating print time (even 30 seconds on 24 core cpu ...).

On a screenshot you can see, that fully loaded were just 2 cores in second row.
[image]

This is feature request.

#### Project File (.3MF) where problem occurs
I cannot upload models with which I am having problem due to license of those models. But it is any complex model.

rtyr, 2021-10-30:
You may try https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-beta1.

G-code export was parallelized and optimized. The total slicing time is 2x to 4x shorter.

vladimir-aubrecht, 2021-11-09:
Thank you, tried it and it's super fast now! :)

closed 2021-11-09 by vladimir-aubrecht