prusa3d/PrusaSlicer #7207: Display of object labels does not match with the "View"-menu, if "Complete individual objects" is used

stefan-am, 2021-10-30:
### Version
PrusaSlicer
Version:   2.4.0-beta1+win64
Build:     PrusaSlicer-2.4.0-beta1+win64-202110251333

### Operating system type + version
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
Create a new project, add a box 19 mm high, add some instances of the box as in 2nd picture. 
Select output option "Complete individual objects"  (see pic1) and let the slicer arrange the objects.  
Object labels are shown despite "Show labels" is not selected in the "View" menu (pic2)
Selecting "show labels" makes the labels disappear (pic3)

_Is this a new feature request?_NO

[image]

[image]

[image]

fe60, 2021-11-09:
I'm not able to reproduce the issue using 2.4.0-beta1 on a Windows machine.
This is what I did:
- Added the box,
- reduced size,
- added three additional instances pressing `+`,
- switched to `Print Settings", activated the option `Complete individual objects`,
- switched back to `Plater` and activated labels pressing `E`.

Result: Labels are shown and the related entry in the `View` menu is checked.

Can you add a project file of the setup you reproduced the issue with?
When and how did you activate the visualization of the labels?

lukasmatena, 2021-11-09:
I did not reproduce it either. Can you reproduce it every time? Can you provide exact sequence of steps since the start of PrusaSlicer that leads to this? Thanks.

stefan-am, 2021-11-09:
I had some difficulty to reproduce this on my own... 
One command sequence i can reproduce it is:

Close Slicer and start it new.
Open the "Label_Shape-Box.3mf" 
Select  View>Collapse Sidebar
Select  View>Full Screen
Press "E" to show labels. Labels are shown on the Plater but in the View Menu it is not checked
[image]
.
[Label_Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/7506009/Label_Shape-Box.zip)

fe60, 2021-11-10:
Reproducing the issue depends on the experimental GUI preference _Set settings tabs as menu items_.
As soon as this is active, the issue occurs.

Just add one object, press `E` and check the _View_ menu.

rtyr, 2021-12-02:
> Reproducing the issue depends on the experimental GUI preference _Set settings tabs as menu items_. As soon as this is active, the issue occurs.
> 
> Just add one object, press `E` and check the _View_ menu.

I was able to reproduce this on current master.

FidelCapo, 2021-12-03:
This issue should be fixed in the next release.
Thank you for the repro.
Closing.

closed 2021-12-03 by FidelCapo