prusa3d/PrusaSlicer #7208: Unable to update due to an inability to connect to COM port

GioGM, 2021-10-30:
### Version
2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3 / prusa3d_fw_3_10_0_MK3
### Behavior
* Unable to update firmware
* 1. Download latest firmware.
* 2. follow usual steps to install latest firmware
 

* _Expected Results_
* firmware updates
* _Actual Results_
  * flashing fails and I get this in the output log:
  *  * avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\G****\Downloads\prusa3d_fw_3_10_0_MK3\prusa3d_fw_MK3_3_10_0_4481.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Jul 16 2021 at 12:28:18
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
avrdude-slic3r: stk500v2_recv(): timeout
avrdude-slic3r: stk500v2_getsync(): timeout communicating with programmer
avrdude-slic3r: Could not open port: COM3

avrdude-slic3r done.  Thank you.

https://user-images.githubusercontent.com/48390677/139560875-52afaed7-6110-4d45-8320-14ae2323b2bd.mp4

NOTE
 i have tried, fac reset, reinstall drivers and slicer, tried another usb cabel, change com port, another computer and upload older firmware, but all make the same error
already tried all these methods with no success
https://www.reddit.com/r/Artillery3D/comments/gylc7x/firmware_update_error_timeout_communicating_with/

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/firmware-update-issue-avrdude-has-no-access-to-com-port/

https://github.com/prusa3d/PrusaSlicer/issues/2662

https://www.help.prusa3d.com/en/article/firmware-updating-and-flashing-problems_2225

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-11-01:
>       1. follow usual steps to install latest firmware

Don't you have Cura or any other program running?

GioGM, 2021-11-02:
I deleted Cura from my PC before trying to flash the firmware

husqy-701, 2022-01-29:
@GioGM 
Having exactly the same issue with my prusa mk3s clone...
mine lcd is stuck on "Original prusa i3 prusa research ..."

did you manage to make it work?

closed 2026-09-29 by prusaslicer-bot[bot]