prusa3d/PrusaSlicer #7209: "Avoid crossing perimeters", "Seam position Nearest" and "External perimeters not first" not working when thin walls detected

dartrax, 2021-10-31:
### Version
2.4.0 beta 1

### Operating system type + version
Win 10

### Behavior
[image]_Describe the problem_
When Detect thin walls enabled, Avoid crossing perimeters and nearest seam position are not working as expected. In my example, in every layer first the thin walls are printed. From there, the print head travels to a position where the first seam is generated. Three issues:
- This travel path does not avoid perimeters, it crosses a hole while it could perfectly travel around that hole (and does if detect thin walls is disabled).
- The perimeter that gets printed first is an outer perimeter, but "External perimeters first" is not checked.
- Also, the placement of this seam is not "nearest", after the perimeter is finished, the toolhead travels back from where it came from, the area with the thin walls - nevertheless, now it avoids perimeters.

* _Steps needed to reproduce the problem_
Slice the attached 3mf files
* _Expected Results_
[image]
_Actual Results_
[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[DetectThinWalls.zip](https://github.com/prusa3d/PrusaSlicer/files/7448631/DetectThinWalls.zip)

JasonGuthrie, 2021-11-01:
I believe this is somewhat related to the excessive travel movements I was having in issue #7219.

hejllukas, 2021-12-02:
There was an issue in avoid crossing perimeters, when "External perimeter first" or "Detect thin walls" was enabled, then travels could lead through holes. This problem has been there since 2.3.0 but only became more common after a seam planning changes.
So at least the issues with avoid crossing perimeters should be fixed with 7f951c2df8e0b654c08c118502c8378c05fde435.

dartrax, 2021-12-02:
Hi @hejllukas,
thanks for the first fix! I've updated my issue description accordingly.

> The issue with "Detect thin walls", that forces the external perimeters to be printed first, and the seam placement issue remains untouched for now.

Okay - a little challenge for later :-)

dartrax, 2021-12-08:
> * The perimeter that gets printed first is an outer perimeter, but "External perimeters first" is not checked (same as in 2.3.0).

I just noticed that this was already reported in #6909.

However, placement of the seam not being the „neatest“ is still valid.

dartrax, 2022-07-06:
I'm closing my report because 2 of 3 issues are fixed and the remaining issue is a duplicate of #6909.

closed 2022-07-06 by dartrax