prusa3d/PrusaSlicer #7210: more extrusion overlap settings and "fill gaps" information/settings/disambiguation!

nubnubbud, 2021-10-31:
This is a feature request, and probably a rather unpopular one, because it's technically not useful?... but all these settings have caused me quite a bit of pain in the past because of their interconnectedness.
but it is good for troubleshooting and hacky/unusual prints at the very least.

Simply put, I noticed that changing extrusion width ALSO changed extrusion paths, to ensure the extrusions were always edge to edge- but being an easily confused engineer- I spent hours trying to get them to overlap in one way or another, but the only setting to adjust their overlap was purely for the transition between infill and perimeter- that wasn't very useful for my purpose.

To top it off, There is NO documentation to suggest that infill, even when 100% solid has the same "fill gaps" feature as perimeters, [this math, which ](https://manual.slic3r.org/advanced/flow-math) I assume it doesn't in my arrogance and lack of knowledge about the codebase.

TL;DR, these 3 features would assist each other greatly.
1. information on exactly what settings affect each other, and additional information on currently unlisted(unexpected) effects of settings, not just what they base their default on, such as extrusion width also affecting perimeter

2. apply "fill gaps" flow calculation to ALL extrusion, if it isn't already. as covered in feature 1, we just don't know if it exists yet, and because feature 3 isn't implemented, we can't test for ourselves!

3. allow the changing of various line overlaps. This includes all perimeters, infill, supports, and top/bottom layers. Ideally, this could manifest as a separation of "extrusion width" and "extrusion spacing". At the moment, we have no way to control how aggressively gaps are filled, and extrusion multiplier is a VERY heavy handed way to do it.
[image]
 maybe something like this? We already have a section for overlap. I put 0's there, though I know, with fill gaps as a standard, they are definitely above 0 for perimeters!

dartrax, 2021-10-31:
Have a look at SuperSlicer. It's a fork of PrusaSlicer and does exactly what you want.
[image]

neophyl, 2021-10-31:
Susi may display them and let you set either but it isn't actually uncoupled with respect to the extrusion width and the extrusion spacing.  If you change the spacing the width is adjusted and vice versa.

dartrax, 2021-10-31:
Yes, you need to change the overlapping. So this is only related to your point no. 3, and the Help/Details section in SuSi could be what you described in point no. 1.

nubnubbud, 2021-11-01:
that's a pretty good way to manage it, and the help and details section would be fantastic.

however, even that one, in the lower overlap area, there's no way to set the overlap of two infill lines. I'm mostly doing this to enable clear printing without overextrusion. This requires some pretty bonkers precision, not gonna lie, but it definitely can't be done if you have no granular control over the overlap of every line type, or at least the information of which line types get that treatment in the code.it seems most do, but not support material, which is what I feared- that the overlap setting isn't just a global setting all line types have.

@dartrax it's still a little confusing to me, though. the spacing is denoted as the width of blocked extrusion edges... is it actually supposed to be from extrusion center to the next extrusion center? that would make far more sense as a measurement - it's very odd to measure using the length of something determined by its surroundings...

closed 2026-09-29 by prusaslicer-bot[bot]