prusa3d/PrusaSlicer #7212: [2.4.0-beta1] Crash when uploading to FlashAir

vintagepc, 2021-10-31:
### Version
PrusaSlicer
Version:   2.4.0-beta1+linux-x64-GTK3
Build:     PrusaSlicer-2.4.0-beta1+linux-x64-GTK3-202110251340

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.3.18-59.19-default x86_64
Total RAM size [MB]: 33,611MB
OpenGL installation
GL version:   4.6.0 NVIDIA 470.57.02
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1050/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<

### Behavior
I have a MK3S with a (genuine) FlashAir SD card. I've been uploading without issue in the alpha3 and in the prior release. 

However, the beta will arbitrarily hard-crash during the file upload. Usually the first one or two slices and uploads will go fine, but then the third one will hard crash and the file will only be partially transferred. 

I've started it from the commandline in the hopes of catching a more detailed exception message and will update if I get any additional information. 

_Is this a new feature request?_
N/A - I've had this happen with three completely different sets of STLs; sometimes when changing settings and re-slicing the same plate so it does not appear to be related to a specific model or configuration.

lukasmatena, 2021-10-31:
Sounds like a duplicate of #7170, which was already fixed.

vintagepc, 2021-10-31:
Excellent, quite possible since I imagine there's a lot of overlap between the two functionalities :)

bubnikv, 2021-12-08:
Hopefully fixed with #7170, closing.

closed 2021-12-08 by bubnikv