prusa3d/PrusaSlicer #7215: Infill is placed before perimeters.

fjpanag, 2021-10-31:
### Version
Version 2.3.0+linux-x64-GTK3

### Operating system type + version
Fedora 34

### 3D printer brand / version + firmware version (if known)
Custom build delta, based on Duet3D board.

### Behavior
`Infill before perimeters` is **not** selected, however Prusa Slicer does place infill first.

See the attached project where the issue is reproduced.

#### Project File (.3MF) where problem occurs
*Note, `.3mf` is renamed to `.zip` for uploading.*

[dog_love_dog.zip](https://github.com/prusa3d/PrusaSlicer/files/7449420/dog_love_dog.zip)

neophyl, 2021-10-31:
OK that's a weird one.  If I load your project then slice, yes it does do the infill first before perimeters.  If I then just turn off Detect Thin walls and re-slice it then correctly does the perimeters first and then infill.  
Turning detect thin wall back on afterwards however does not make it go back to infill first and so seems to 'fix' whatever was causing it.

dartrax, 2021-10-31:
Confirmed, and this bug is also present in 2.4.0 beta 1.

> OK that's a weird one. If I load your project then slice, yes it does do the infill first before perimeters. If I then just turn off Detect Thin walls and re-slice it then correctly does the [...]
I can't reproduce this behavior, nor in 2.3.0 x64 (Windows) neither in 2.4.0 beta 1 (Windows). I always see consistent (wrong) behavior here.

bubnikv, 2021-11-01:
This is an old error, which is not new to PrusaSlicer 2.4.0-beta3. I tested with PrusaSlicer 2.3.0 with the same results.

This issue seems to be related to the particular model and "detect thin walls" enabled.

bubnikv, 2021-11-01:
PrusaSlicer 2.1.0 slices this project correctly, PrusaSlicer 2.2.0 incorrectly.

bubnikv, 2021-11-01:
Fixing by Netfabb fixes this issue.

FidelCapo, 2025-04-14:
> This can be closed as fixed?

I think so. Thank you.

closed 2025-04-14 by FidelCapo