prusa3d/PrusaSlicer #7217: Frozen after waking windows from sleep

jtaseff, 2021-10-31:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10 pro 21H1

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Consistent crash on this PC. After importing and slicing any STL file, leaving it in layer preview, the PC goes to sleep automatically. Upon waking it, PrusaSlicer is always frozen on the preview screen and the application goes dim. It uses a ton of CPU for a long time but never becomes responsive.

#### Project File (.3MF) where problem occurs
Happens consistently with any STL file loaded

dartrax, 2021-11-01:
Can't reproduce with 2.3.0 or 2.4.0 beta 1 on NVIDIA GeForce MX150, may be hardware dependent.
By "Sleep" do you mean hibernating or power save?
Could you try with 2.4.0 beta 1?
What's your GUI Setting (Old regular layout, New layout, non-modal window)?
What's your graphics card?

bubnikv, 2022-01-03:
I have experienced a hang-up or broken OpenGL rendering on my laptop, but it is not reproducible and it happens rarely. I am not sure whether it used to happen for 2.3.3, I only remember seeing it with 2.4.0 and  2.4.0 pre production builds.

JPA1238, 2024-06-23:
I am currently also experiencing this issue. Running prusaslicer 2.7.4 on windows 11. When my laptop goes to sleep, the bug happens and the part of the window with the buildplate does not update anymore. I can still go to the different settings tab and edit objects on the right side. Slicing also works no problem. Just visually nothing updates anymore on the buildplate window. After slicing it also doesn't switch to the sliced view. 
When hibernating, this doesn't happen.

closed 2026-09-29 by prusaslicer-bot[bot]