prusa3d/PrusaSlicer #7218: [CZ] BUG v 2.4.0 alpha

karlost, 2021-10-31:
### Version
alpha 2.4.0

### Operating system type + version
win 10 last update

V pokud težím pres NBminer , prusa slicer padá hned při startu. bez hlašky, 
v momentě kdy vypnu těžbu, normálně funguje, zapnu těžbu a jako by to prusaslicer zral 99% GPU ,, z 30mh/s se drží na 500kh/s 

považovat za bug nebo hold se nedá slicovat při miningu ?

dartrax, 2021-11-01:
DeepL translation:

> If I try to use NBminer , prusa slicer crashes right at startup. without the header,
> the moment I turn off mining, it works normally, turn on mining and it's like the prusaslicer eats 99% of the GPU, from 30mph/s it stays at 500kh/s
> 
> Consider it a bug or hold it can't be sliced while mining ?

bubnikv, 2021-12-08:
Nedá se slajsovat při majningu.

closed 2021-12-08 by bubnikv

dartrax, 2021-12-08:
> Nedá se slajsovat při majningu.

DeepL translation:
You can't slack off when you're pitching.

??? What does that mean?