prusa3d/PrusaSlicer #7219: Bug: Excess travel moves on circular parts

JasonGuthrie, 2021-11-01:
### Version
PS 2.4.0 Beta1
### Operating system type + version
W10

### 3D printer brand / version + firmware version (if known)
i3 Mk3

### Behavior
When printing a circular part with minimal perimeters (tested with 1 and 2), there is a significant amount of excess travel to the next seam position (in this example a full lap is made on every single layer). I have tried all variations of seam positioning with and without 'avoid crossing perimeters' on but nothing seems to make any difference.

This is wreaking havoc with this particular part (a bellows for a coffee grinder), as it be being printed with soft TPE. The excess travel move causes the extruder to leak over half of the part, causing blobbing. This is then exacerbated by the underextrusion once the next layer is started at the new seam position. 

I cross checked with PS 2.3 and the latest release of SuperSlicer and this issue was not present. 

This may be related to #7209

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
[MC7 Bellows.zip](https://github.com/prusa3d/PrusaSlicer/files/7450079/MC7.Bellows.zip)
[image]

dartrax, 2021-11-01:
I don't think it's related to thin wall detection. What we see here is not a travel move, but an excessive Wipe move, wiping all around that part...
[image]
I'm not sure if there is any setting in your profile causing this. Seems more like a bug to me.

JasonGuthrie, 2021-11-02:
@dartrax you are 100% right, it is the wipe command that is causing all the issues. Well spotted!

I have been having issues with it attempting to iron half the part (as you say) before but I never realise it was the wipe that was causing the issue. I will have to keep digging to see what the issue is although it very well may be a bug.

JasonGuthrie, 2021-11-02:
@dartrax I found the problem. In my custom flex filament profile I had forgotten to turn off retraction via the length value. Once I had set this to zero the problem disappeared completely. 

[image]
[image]

closed 2021-11-02 by JasonGuthrie

dartrax, 2021-11-02:
> @dartrax I found the problem. In my custom flex filament profile I had forgotten to turn off retraction via the length value. Once I had set this to zero the problem disappeared completely.

That's not the root cause for that behavior. I found something else.

Short story:
Your Travel Speed is set to 8000mm/s (!). This influences calculation of the wipe length. Lower that value and you can actually use the wipe setting, or lower the retraction length to very small (0.05mm) values.

Long story:
And by setting the _retraction_ length to small values, you can actually influence the _wipe_ length. I didn't know that!
Retraction length is expressed in raw filament length, so a small value here leads to a huge value for the travel length.
In this case, 0.1mm retraction length will result in about 38mm wipe. And in this case, you can actually see the E parameter doing the retraction... Yes, because it's the "Wipe _while_ retracting" setting!

How to control the speed of the retraction and therefore, the length of the wipe? (I think this should be better documented (#4655)). The retraction speed is set to 35mm/s. The wipe speed should be somewhere between print (30mm/s) and travel 
 (8000 (!) mm/s) speed. In [this comment](https://github.com/prusa3d/PrusaSlicer/issues/2507#issuecomment-678590809) it is said that it is travel speed, in your GCode, travel speed is F480000, wipe speed is F384000. So I assume it takes the time it needs to retract the amount of filament set by "retraction length" with the "retraction speed" and calculates how long it can travel using the wipe speed in this time span. Since your travel speed is very high, this results in a very long wipe.