prusa3d/PrusaSlicer #7220: Non Responsive & Memory Leak when opening or changing this 3MF project

FlashTech, 2021-11-01:
### Version
2.4.0-beta1+win64

### Operating system type + version
PrusaSlicer
Version:   2.4.0-beta1+win64
Build:     PrusaSlicer-2.4.0-beta1+win64-202110251333

CPU: Ryzen 9 5900X

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,300MB
OpenGL installation
GL version:   4.6.0 NVIDIA 471.41
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3080 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 343 lines cut ...]
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3s+ MMU2s

### Behavior
The 3mf file I have freezes for 30 seconds or a couple minutes and the RAM usage skyrockets to many GB when I open the project or anytime I change any objects and changing any print settings. For example, if I change the infill from 15% to 10% it freezes (see screenshot). I was able to use the 3mf, make modifications and export the gcode. The problem is it took an hour repeatedly making small changes and waiting for it to respond again and again. I successfully printed the gcode too on my printer. Also opening this 3mf in a previous PrusaSlicer version did not experience this problem!

To reproduce simply open the 3mf file and watch your RAM and how long it takes for the program to respond changing any parameter, like infill %.

#### Project File (.3MF) where problem occurs (zipped because .3mf was not allowed to be uploaded.
[12xResWheat_v06.zip](https://github.com/prusa3d/PrusaSlicer/files/7450568/12xResWheat_v06.zip)

[image]
[image]

dartrax, 2021-11-01:
Confirmed.

bubnikv, 2021-11-01:
I see it as well. Weird. I am looking into it.

bubnikv, 2021-11-01:
Fixed with a0ee41770d390efb7c298c397172e1198c85f08d

    Fix of Print::apply() creating hierarchy of regions for complex scenarios:
    Unnecessary regions were created for a modifier over a volume or a modifier,
    where the modifier did not modify any of its parent's properties.
    This lead to an explosion of regions for this particular 3MF.
    Fixes Non Responsive & Memory Leak when opening or changing this 3MF project #7220

closed 2021-11-01 by bubnikv