prusa3d/PrusaSlicer #7222: autocloses after slicing and sending to ocotprint

swasti2022, 2021-11-01:
### Version
_Version of PrusaSlicer used goes here_
Version 2.4.0-beta1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
WIN10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
prusa mk3s+

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
autocloses after slicing and sending to ocotprint
i had this problem now 3 or 4 times
noting special sliced
after dirct sending from prusaslicer to octoprint, ps closes

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

dartrax, 2021-11-01:
This is probably a duplicate of #7170, which is already fixed.

lukasmatena, 2021-11-01:
Thanks for the report, but @dartrax is right. Closing.

closed 2021-11-01 by lukasmatena

swasti2022, 2021-11-01:
Ah. Sorry... didnt see that there was a similar report.Regards