prusa3d/PrusaSlicer #7223: erreur avec la tour de nettoyage (use_relative_e_distances=1)

18pool49, 2021-11-01:
### Version 2.3.3. et 2.3.2
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version W10  version 21H1 (19043.1288)
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) "Tenlog TL-D3PRO marlin d'origine "
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior  :::: erreur (use_relative_e_distances=1) 
* _Describe the problem_ quanhd "je coche la tour l'erreur est présente 
[image]
[image]

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs 
[coeur_9 en 36coupes 2 coul prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/7452303/coeur_9.en.36coupes.2.coul.prusa.zip)

_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-11-02:
The wipe tower is currently supported only for relative extruder positioning (can be enabled in printer settings). Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3626. Closing.

closed 2021-11-02 by rtyr

18pool49, 2021-11-02:
bon j'ai résolu a l'écran ce problème de tour qui bloquait la découpe 
j'ai coché  dans >>réglages imp. >>général>> avancé  -----> .utiliser des valeurs E relatives [v]
et le découpage a l'écran se fait correctement 
! ! !il me reste a vérifier si le Gcode est correcte pour le "Marlin" de la "Tenlog"

18pool49, 2021-11-02:
[image]