prusa3d/PrusaSlicer #7224: Excessive M204 Commands in thin walls

dartrax, 2021-11-01:
### Version
2.4.0 beta 1

### Operating system type + version
Win 10

### Behavior
* _Describe the problem_
When thin walls are printed, every segment (of different width) is preceeded by
`M204 P800
M204 P400`
Not only the first command has no effect as it gets overridden by the second, it should only be necessary to write it only once at the beginning of the thin wall.
* _Steps needed to reproduce the problem_
Slice the attached .3mf and look at Line 140.
* _Actual Results_
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[M204.zip](https://github.com/prusa3d/PrusaSlicer/files/7453884/M204.zip)

bubnikv, 2021-11-01:
I suppose this is how it always worked. I don't think we will touch this code for PrusaSlicer 2.4.0 release.

Currently the acceleration is overridden for extruding moves. What you see is resetting the acceleration to its default (M204 P800) at the end of each extrusion move. We need to analyze the effects of postponing resetting the acceleration to travel moves or any other possible moves, also we need to analyze the effects of postponing resetting acceleration to its default before any custom G-code or possibly not doing it at all for custom G-codes.

dartrax, 2021-11-01:
That's ok as it doesn't hurt. Only Projects with many thin walls may benefit from smaller files, and possibly faster slicing (?), and I agree that there are more important bugs waiting to be fixed for the release.

Riksu9000, 2022-01-15:
This actually doesn't happen in version 2.3.3 for me, so I think it's a new issue with 2.4.0, and it's causing my printer to freeze... It's running Klipper with a macro that sets accel_to_decel to half the accel set by M204. Either disabling the macro or using PS 2.3.3 fixes the freezing.

The excessive M204s are more likely to happen on the first layer for some reason. All extrusions in these screenshots should be the same width, so it's not related to first layer extrusion width. These screenshots show the first and the second layer.

[image]
[image]

lukasmatena, 2022-01-17:
> it's causing my printer to freeze...

I believe that setting acceleration twice in a row is still a valid G-Code. I sliced your 3MF in 2.3.3 and the double M204 is also there. I tried in 2.0.0 (!) and it is there also. This is not new.

Riksu9000, 2022-01-17:
I didn't provide any 3mf, but I can do that now. Here's a 3mf file and gcodes I've sliced. With 2.3.3 there are 14 double M204s and 2.4.1 there are 902, so there is a difference, and it's enough to cause the issues I've mentioned.

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/7882455/test.zip)

closed 2026-09-29 by prusaslicer-bot[bot]