prusa3d/PrusaSlicer #7225: Randomizing perimeter line start/end position within the same layer

laukejas, 2021-11-01:
### Version

Version 2.3.3+win64

### Operating system type + version

Windows 10 x64

### 3D printer brand / version + firmware version (if known)

Prusa Mk3S (irrelevant)

### Behavior

Currently, PrusaSlicer does not allow to randomize the seam position in the same layer, meaning that with most prints, every perimeter line starts and ends at the same location, and these "gaps" line up. This causes a lot of issues when trying to produce a watertight print, because this gap provides a straight path for the water to get to the inside of the model.

[image]

It would be great if there was an option in the slicer settings to randomize the position of the start and end of each perimeter line, so they don't line up like this. It would greatly increase the watertightness of the 3D prints.

#### Project File (.3MF) where problem occurs

[pressure test_Default.zip](https://github.com/prusa3d/PrusaSlicer/files/7455952/pressure.test_Default.zip)

rtyr, 2021-11-02:
You may try PrusaSlier 2.4.0-beta1, where the seams between the nested perimeters are not aligned exactly, but they are staggered by a short distance to reduce bulge of the outer most seam.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-beta1

[image]

vincele, 2022-03-04:
This looks like a good first step. I'd go even further and ask for "maximized distance between adjacent perimeters". For water tightness, Adjacent may also be on the Z axis.

closed 2026-09-29 by prusaslicer-bot[bot]