prusa3d/PrusaSlicer #7226: Feature Request: Add option dwell + reprime between fast layers (alternative to slowing down print speed)

FIRES-3D, 2021-11-01:
### Version
PrusaSlicer 2.3.3+win64

### Operating system type + version
Windows 10 Pro 64-bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Feature Request Description
Add options in "Filament Settings -> Cooling -> Cooling Thresholds" box: 
- "Dwell and reprime nozzle between layers if layer print time is below _______ seconds." (Default = 0 / Disabled)
- "Minimum repriming length: ______ mm" 

This feature would permit normal print speeds to always be maintained regardless of layer size. After a fast layer is printed, the remaining layer time would be spent with the nozzle dwelling off to the side, then re-priming using a short line, then resuming the next layer. 

Note, for best results the repriming feature/geometry should be printed with every single layer even when it's not needed (perhaps print a circle instead of a line/wall to avoid issues with tall prints). It can be quite small so should not be excessively wasteful (and, this option would likely be disabled by default anyway)

### Reasoning
Reducing nozzle speed to extend layer times is not a perfect solution in every case. Below are some potential undesirable outcomes:
- Inconsistent surface finish (sheen) due to varying perimeter speeds between layers
- Inconsistent dimensional accuracy due to varying perimeter speeds between layers (can feel tiny "ridge" at each fast/slow layer transition)
- Curling due to very hot nozzle spending extended time in close proximity to freshly-extruded layer (potentially also blocking cooling airstream)
- Inconsistent filament extrusion temperature; i.e., the temperature of the actual filament as it exits the nozzle. Lower print speed = more heating time = higher filament temp (regardless of nominal heatblock temp setting)
- Poor performance on very small layers. For various reasons, a reasonable "min print speed" is desirable, but this means very small layers are unable to be sufficiently cooled. 

Adding an option to dwell + reprime nozzle would enable the entire print to be completed (consistently) at the nominal settings, while also providing sufficient cooling time between layers.

Note, this suggestion applies particularly to:
- Fast prints (90+ mm/s), where the difference between nominal print speed and min print speed is very large
- Wide nozzles (0.6mm), for which the larger layers require additional cooling, slow nozzle speeds can increase filament oozing, and filament/heatblock temp differences are accentuated (especially at high print speeds).

dartrax, 2021-11-02:
This idea also evolved in [#6136](https://github.com/prusa3d/PrusaSlicer/issues/6136#issuecomment-786580331).

bubnikv, 2021-12-08:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7177
Closing this one.

closed 2021-12-08 by bubnikv

batata004, 2021-12-20:
@dartrax I think you provided the best solution in all the threads about this subject! Instead of creating a tower, instead of waiting and retracting filament... what about increasing the infill area on layers where the printing time is below a threshold? I think this is awesome cause at least the "extra" filament is being printed on some useful area (instead on a tower). And I think this solution of yours should be pretty easy to implement!