prusa3d/PrusaSlicer #7228: réduire volume de la tour

18pool49, 2021-11-02:
### Version
_Version of PrusaSlicer used goes here_ 2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version W10 
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) tenlog TL-D3PRO      Marlin 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_  la tour a toujours une surface de 90mm2  comment réduire cette surface 
* 
[image]
[image]
[image] largeur tour >>20 mm
( 30x30 ou 20x45 ou ??x??=  TOUJOURS >>> 80 a 90 mm2) = 2100 mm de dépose a chaque changement de couleur  = 480 mm de long de fil c'est énorme 
[image] largeur tour >> 10 mm

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-11-02:
Reduce the purge volumes.

[image]

Closing.

closed 2021-11-02 by rtyr

18pool49, 2021-11-02:
milles merci de m'ouvrir les yeux 
mais a l'écran cette fonction est pas dés plus visible (pas complète )