prusa3d/PrusaSlicer #7229: Bug: Missing first layer when having intersecting supports and brim

martingwerder, 2021-11-02:
### Version
Prusa Slicer 2.4.0-beta1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MMU2S/3.10.0+1.0.6

### Behavior
* When rendering a model with inner brim and support structures the first layer of the support structure on the overlapping area is missing.
* Create a brim and support structures to be set on the same location.
* First layer should be a solid brim/support layer
* but on intersecting areas the first layer is missing
  * Screenshot of an area *without* inner brim (slicing as expected) [image]
  * Screenshot of an area with inner brim (1st layer only and first two layers) [image] [image]

No feature request. It is a Bug report

#### Project File (.3MF) where problem occurs
[Whirlwind_2.4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7460182/Whirlwind_2.4.3mf.zip)

hejllukas, 2021-11-02:
It should be fixed with commit 30a37d0c2583fcc93bef7922d38dfc055ee1c13b.
[image]
[image]
There was an issue (#7213) that the inner brim wasn't generated, and I was also causing your issue.

closed 2021-11-02 by hejllukas

martingwerder, 2021-11-03:
Thanks for the quick reaction. I look forward to the next beta release.