prusa3d/PrusaSlicer #7231: Crash after start-up

zaitcev, 2021-11-02:
### Version
[zaitcev@suzdal 3d_print]$ rpm -qf /usr/bin/prusa-slicer
prusa-slicer-2.3.1-1.fc34.x86_64

_Use `About->About PrusaSlicer` for release versions_
Impossible to collect About->PrusaSlicer because it crashes.

### Operating system type + version
Fedora 34

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3s+

### Behavior
The PrusaSlicer starts, displays the main window and then immediately crashes.

The following is printed to the terminal:

................ many GTK warnings
(prusa-slicer.wrapped:61292): Gtk-WARNING **: 11:26:52.888: Negative content height -9 (allocation 1, extents 5x5) while allocating gadget (node button, owner GtkButton)

(prusa-slicer.wrapped:61292): Gtk-CRITICAL **: 11:26:52.888: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
11:26:53 AM: Debug: window wxTreeCtrl(0x557a723367d0, ) lost focus even though it didn't have it
11:26:53 AM: Debug: window wxTreeCtrl(0x557a723367d0, ) lost focus even though it didn't have it
/usr/include/c++/11/bits/basic_string.h:1057: std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::reference std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::operator[](std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::size_type) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>; std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::reference = char&; std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::size_type = long unsigned int]: Assertion '__pos <= size()' failed.
Aborted (core dumped)

#### Project File (.3MF) where problem occurs
None

combolek, 2021-11-03:
I am on Fedora 34 and I cannot reproduce it with the exact same package (presumably from here? https://kojipkgs.fedoraproject.org/packages/prusa-slicer/2.3.1/1.fc34/x86_64/prusa-slicer-2.3.1-1.fc34.x86_64.rpm). Note though that this package hasn't been promoted to stable (which is still version 2.2).
[image].

zaitcev, 2021-11-03:
Indeed it didn't start happening outright. I managed to make a gcode once. Then I exited the prusa-slicer, looked at some STLs, and literally on the next start-up it started crashing like that.

zaitcev, 2021-11-03:
I'll be happy to look deeper, build from source, etc., but I need some hand-holding.

combolek, 2021-11-03:
Did you try `PrusaSlicer-2.3.3+linux-x64-GTK3-202107161044.AppImage` ?

Also, if it worked before then maybe this is something in your settings. Try deleting (or renaming out of the way) your profile directory `~/.config/PrusaSlicer`.

lukasmatena, 2021-11-04:
@zaitcev 
> Try deleting (or renaming out of the way) your profile directory

The renaming is highly preferred. If the problem is indeed in the config directory (which I doubt in this case, but who knows), it can be investigated.

zaitcev, 2021-11-05:
I did the following:
1. launched prusa-slicer 2.3.1 to make sure it still crashes
2. mv .config/PrusaSlicer/ dot.config.PrusaSlicer
3. launched prusa-slicer, it went to configuration wizard
4. went through wizard, everything seemed in order
5. exited
6. launched again and it crashed as before

zaitcev, 2021-11-05:
The systemd steals the core, but this is what I got with gdb:

(gdb) where
#0  0x00007ffff5b292a2 in raise () at /lib64/libc.so.6
#1  0x00007ffff5b128a4 in abort () at /lib64/libc.so.6
#2  0x000055555573ab98 in std::__replacement_assert(char const*, int, char const
*, char const*)
    (__file=<optimized out>, __line=<optimized out>, __function=<optimized out>,
 __condition=<optimized out>)
    at /usr/include/c++/11/x86_64-redhat-linux/bits/c++config.h:2654
#3  0x0000555555786769 in std::__cxx11::basic_string<char, std::char_traits<char
>, std::allocator<char> >::operator[](unsigned long) [clone .part.0] [clone .lto
_priv.0] (__pos=<optimized out>, this=<optimized out>)
    at /usr/include/c++/11/bits/basic_string.h:1057
#4  0x0000555555c16aa5 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long)
    (__pos=<optimized out>, this=0x555558aadd40, this=<optimized out>, __pos=<optimized out>) at /usr/include/c++/11/bits/basic_string.h:1057
#5  Slic3r::GUI::NotificationManager::PopNotification::render_text(Slic3r::GUI::ImGuiWrapper&, float, float, float, float)
    (this=0x555558aadc90, imgui=..., win_size_x=<optimized out>, win_size_y=<optimized out>, win_pos_x=<optimized out>, win_pos_y=<optimized out>)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/NotificationManager.cpp:516
#6  0x0000555555c232c4 in Slic3r::GUI::NotificationManager::PopNotification::render(Slic3r::GUI::GLCanvas3D&, float, bool, float)
--Type <RET> for more, q to quit, c to continue without paging--
    (this=0x555558aadc90, canvas=<optimized out>, initial_y=<optimized out>, move_from_overlay=<optimized out>, overlay_width=<optimized out>)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/NotificationManager.cpp:228
#7  0x0000555555c84eaf in Slic3r::GUI::NotificationManager::render_notifications(float) (overlay_width=0, this=0x555557025de0)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/NotificationManager.cpp:1255
#8  Slic3r::GUI::GLCanvas3D::render() (this=0x555557339a20)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/GLCanvas3D.cpp:1748
#9  0x0000555555c8d101 in Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&)
    (evt=..., this=0x555557339a20)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/GLCanvas3D.cpp:2441
#10 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&)
    (this=0x555557339a20, evt=...)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/GLCanvas3D.cpp:2412
#11 0x00007ffff72c15c3 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#12 0x00007ffff72c37f2 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
    at /lib64/libwx_baseu-3.0.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#13 0x00007ffff72c3884 in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /lib64/libwx_baseu-3.0.so.0
#14 0x00007ffff72c393f in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /lib64/libwx_baseu-3.0.so.0
#15 0x00007ffff72c3a31 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /lib64/libwx_baseu-3.0.so.0
#16 0x00007ffff72c17eb in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
    at /lib64/libwx_baseu-3.0.so.0
#17 0x00007ffff6f60a3f in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff6f60a10 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#19 0x00007ffff6f60a10 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#20 0x00007ffff6f60a10 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#21 0x00007ffff6f60a10 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#22 0x00007ffff6e2ea03 in wxFrame::SendIdleEvents(wxIdleEvent&) ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#23 0x00007ffff6e7223d in wxAppBase::ProcessIdle() ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#24 0x00007ffff6dbe745 in wxApp::DoIdle() ()
--Type <RET> for more, q to quit, c to continue without paging--
    at /lib64/libwx_gtk3u_core-3.0.so.0
#25 0x00007ffff6dbe857 in wxapp_idle_callback ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#26 0x00007ffff5e8c74b in g_idle_dispatch () at /lib64/libglib-2.0.so.0
#27 0x00007ffff5e904cf in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#28 0x00007ffff5ee44f8 in g_main_context_iterate.constprop ()
    at /lib64/libglib-2.0.so.0
#29 0x00007ffff5e8fa93 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#30 0x00007ffff621ae1d in gtk_main () at /lib64/libgtk-3.so.0
#31 0x00007ffff6dd3475 in wxGUIEventLoop::DoRun() ()
    at /lib64/libwx_gtk3u_core-3.0.so.0
#32 0x00007ffff71f25e1 in wxEventLoopBase::Run() ()
    at /lib64/libwx_baseu-3.0.so.0
#33 0x00007ffff71cceda in wxAppConsoleBase::MainLoop() ()
    at /lib64/libwx_baseu-3.0.so.0
#34 0x00007ffff722e6f0 in wxEntry(int&, wchar_t**) ()
    at /lib64/libwx_baseu-3.0.so.0
#35 0x0000555555a8b47c in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&)
    (params=...)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/slic3r/GUI/GUI_Init.cpp:84
#36 0x0000555555759828 in Slic3r::CLI::run(int, char**)
    (this=0x7fffffffd080, argc=1, argv=0x7fffffffd298)
--Type <RET> for more, q to quit, c to continue without paging--
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/PrusaSlicer.cpp:573
#37 0x000055555570559b in main(int, char**)
    (argc=<optimized out>, argv=<optimized out>)
    at /usr/src/debug/prusa-slicer-2.3.1-1.fc34.x86_64/src/PrusaSlicer.cpp:777
(gdb)

bubnikv, 2021-11-05:
Would you please test with PrusaSlicer 2.4.0-beta1? We are going to release 2.4.0 soon, thus it is not worth investigating issues of old builds.
https://github.com/prusa3d/PrusaSlicer/releases

johnsonm, 2021-11-15:
I reproduce this problem with PrusaSlicer-2.3.1+Fedora on Fedora 35.

If I delete the .config/PrusaSlicer/cache directory I can successfully run prusa-slicer once, and then the same problem occurs.

If I move .config/PrusaSlicer out of the way, the behavior persists; I still can run prusa-slicer successfully once, so it's not part of my config that is causing the crash.

This behavior does not reproduce with PrusaSlicer-2.3.1+linux-x64-202104161339.AppImage so it might reasonably be associated with the Fedora build of prusa-slicer, which means that a bug report to Fedora in bugzilla is probably appropriate.

This behavior also does not reproduce for me with PrusaSlicer-2.4.0-beta1+linux-x64-GTK3-202110251340.AppImage

johnsonm, 2021-11-15:
From  Miro Hrončok in the Fedora bugzilla report I raised:
> Sounds like this only happens when compiled with `-Wp,-D_GLIBCXX_ASSERTIONS`.

That would mean that the testing I did with the different Appimages wouldn't be meaningful, if they aren't actually compiled that way...

hroncok, 2021-11-30:
We can build the Fedora package without `GLIBCXX_ASSERTIONS` but that might only cover up an actuall problem.

bubnikv, 2021-11-30:
We are going to release 2.4.0 final soon, thus we are more interested in fixing the issue there if it still exists. We would be thankful for the reports on the current master.

zaitcev, 2021-11-30:
Unfortunately, I had to replace the computer with high-DPI display urgently, and the current one does not have the 2x scaling. Everything works as expected for me there.

[zaitcev@niphredil ~]$ rpm -q prusa-slicer
prusa-slicer-2.3.3-1.fc35.x86_64

hroncok, 2021-12-01:
I'll try to reproduce this on the current master.

hroncok, 2021-12-01:
It seems it requires an old CGAL version, so no dice for me for now :(

```
CMake Error at src/libslic3r/CMakeLists.txt:291 (find_package):
  Could not find a configuration file for package "CGAL" that is compatible
  with requested version "4.13".
  The following configuration files were considered but not accepted:
    /usr/share/cmake/CGAL/CGALConfig.cmake, version: 5.3
```

bubnikv, 2021-12-08:
@hroncok The CGAL version check was removed. You should be able to recompile.

hroncok, 2021-12-08:
Indeed, thanks

hroncok, 2021-12-08:
FWIW I've been told that this has been fixed in 2.3.3 already, but will need to check

lukasmatena, 2023-02-08:
No response for a long time, likely fixed as of 2.6.0-alpha3. Closing. If anyone still sees it, let us know, we will reopen this.

closed 2023-02-08 by lukasmatena