prusa3d/PrusaSlicer #7232: Fan starts at first layer, even though disabled for first layer.

n8bot, 2021-11-02:
### Version
2.4 beta1

### Operating system type + version
Win 10 64

### Behavior
Option to disable fan for first X (1) layer is enabled, but fan is turned on at beginning of first layer.

I think it could be related to this: https://github.com/prusa3d/PrusaSlicer/commit/32ebfa66e936a67160615aec5f2556fc8956a4d7

#### Project File (.3MF) where problem occurs
Can provide if you are not able to reproduce.

bubnikv, 2021-11-03:
Seems to work just fine

[image]

[image]

rtyr, 2021-11-03:
Could you please provide your 3MF project file?

n8bot, 2021-11-03:
Yes here you go
[fanonlayer1.zip](https://github.com/prusa3d/PrusaSlicer/files/7469122/fanonlayer1.zip)
:

n8bot, 2021-11-03:
Also, as a side note, your screenshot that you posted shows that you have a min fan speed of 100% selected, but the fan is still ramped up from layers 4-7. How does that make sense?

n8bot, 2021-11-03:
I think I just discovered that the problem with the fan on layer 1 is caused when there is a raft present. In that case, the fan is enabled on the first layer (the single raft layer) when it should be disabled. It's also enabled for all other layers. When raft is disabled, the functionality returns correctly: fan disabled on first layer.

rtyr, 2021-11-03:
You are right, I was able to reproduce it and it happens only if a raft is enabled.

> Also, as a side note, your screenshot that you posted shows that you have a min fan speed of 100% selected, but the fan is still ramped up from layers 4-7. How does that make sense?

That's because of "full fan speed at layer" parameter (1-4 fan disabled, layer 5-6 is slower, 7 full speed).

n8bot, 2021-11-03:
> That's because of "full fan speed at layer" parameter (1-4 fan disabled, layer 5-6 is slower, 7 full speed).

I noticed that part. But, it seems very counter-intuitive to be able to specify a minimum fan speed, and yet the fan is easily commanded below that fan speed.

IMO a minimum is literally a minimum. As in, the slowest commanded percentage of speed that the fan will actually spin at.

karlla1220, 2021-11-09:
I had similar issue, especially when raft is enabled.

when disabled for first 4 layer, it works. but it seems ignore that setting when fan is disabled for first 3 layer

Please check my 3mf files. 

[shape-box.zip](https://github.com/prusa3d/PrusaSlicer/files/7507413/shape-box.zip)

closed 2021-12-02 by bubnikv