prusa3d/PrusaSlicer #7234: Crash after sending to Repetier Server

vidi2000, 2021-11-03:
### Version
 PrusaSlicer 2.4.0-beta1

### Operating system type + version
macOS Monterey 12.0.1

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro - Marlin 2.0.9.1

### Behavior
Program crashing after sending G-Code to Repetier Server: usually the first (and sometimes the second) upload to Repetier works fine, then the next one result in a crash (this issue was not present before with any alpha release)

_Is this a new feature request?_ : NO

#### Project File (.3MF) where problem occurs
regardless the object or project

RobeeeJay, 2021-11-03:
Same version and OS, I also get a crash after sending G-Code, but to Octoprint, so I don't think this is specific to the server.

rtyr, 2021-11-03:
Likely same issue as https://github.com/prusa3d/PrusaSlicer/issues/7170. It should be already fixed on master.

lukasmatena, 2021-11-04:
Closing. It is certainly the same as #7170.

closed 2021-11-04 by lukasmatena